mavlink/qgroundcontrol #10328: Software crash

ziggurat-sky, 2022-06-26:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: , Windows 10 
- QGC Version: 4.2.3
- QGC build:  stable
- Flight Controller:  Pixhawk 2.4.8
- Autopilot (with version): PX4 

## Detailed Description
The software works well, but when using the flight tab, click on the option (use multi channel mode selection) The software crashes and I exit the program.
There is the same problem in Android software. I used other versions, I have the same problem and I do not know exactly where the problem comes from.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

dizcza, 2022-07-06:
Same issue. Clicking "use multi channel mode selection" in the radio config menu crashes the app with a segfault.

Ubuntu 20.04
Latest PX4 firmware
Pixhawk 2.4.8
R12DS receiver

```
./QGroundControl.AppImage 
Settings location "/home/dizcza/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/dizcza/Documents/QGroundControl/Logs"
Map Cache in: "/home/dizcza/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: QGCCorePlugin(0x5575128c9730) []
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
"v4.2.0"
_recalcFlightPathSegments homePositionValid false
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters cache match failed /home/dizcza/.config/QGroundControl.org/ParamCache/1_1.v2
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
"3D Robotics" "PX4 FMU v2.x"
FactPanelControllerLog: Missing parameter: "1:RC_ASSIST_TH"
Segmentation fault (core dumped)
```

The last two logs correspond to the issue.

sastel, 2022-09-24:
I am also observing this issue on Ubuntu 20.04 LTS with QGroundControl Version v4.2.3.

paulerikf, 2022-10-17:
Same, Ubuntu 20.04 LTS, QGroundControl 4.2.3.

Note: Does not crash when connected to a 1.10... version of PX4, but does crash when connected to a 1.13.1 version of PX4

userUndefined89, 2023-07-04:
Having same issue on Windows 11 Home. Crashing with PX4. Any resolution since last year?

hzaben1, 2023-07-05:
have the same issue in my mac os

MotoLab, 2023-07-13:
Same thing on Kubuntu 22.04
QGC 4.2.7
Latest stable PX4.

FactPanelControllerLog: Missing parameter: "1:RC_ASSIST_TH"
/tmp/.mount_QGrounEc5JJ7/AppRun: line 7: 109834 Segmentation fault      (core dumped) "${HERE}/QGroundControl" "$@"

patrickelectric, 2023-07-13:
Thanks for reporting, a new version with the fix will be released.

HTRamsey, 2024-05-31:
I don't believe that option exists anymore, please reopen if I am wrong

closed 2024-05-31 by HTRamsey