mavlink/qgroundcontrol #10329: Unknown command problem

ldm0330, 2022-06-27:
Hello, I would like to ask if I updated the latest version of QGC. When I click to create a waypoint during use, it will appear unknown: 84, unknown: 16 and display all parameters. How to solve this problem?

1chick, 2022-07-19:
# same issue.
- "add waypoint" button can not be clicked in planview.
[image]
- unknow waypoint ,command combox is blank 
[image]

# Enviroment:
1.QGC Release v4.2.3
2.win10/android os
3.ardupilot,Rover v4.2.2 firmware

LnAYuTaK, 2022-09-23:
你好，我在version 4.2.0中也遇到这个问题，但是我尝试了上面的解决方法并没有解决这个问题，请问下还有其他需要修改的地方么

Hello, I also encountered this problem in version 4.2.0, but I tried the above solution and failed to solve the problem. Is there anything else that needs to be modified

booo, 2022-11-16:
Could you check if the proposed change fixes your problem? You can find binaries for testing in the pull request and workflow runs: https://github.com/mavlink/qgroundcontrol/pull/10491

booo, 2022-12-10:
Fix merged into master.

closed 2022-12-10 by booo