mavlink/qgroundcontrol #10333: QGC build fails with "error: expected type-specifier before QTcpSocket"

ashishmagar600, 2022-07-01:
US: Ubuntu 18.04 LTS

Qt : 5.15.2 (minimal built from source)

QGC : stable_V4.2

from `qgroundcontrol/build`

```
$> qmake ..

Project MESSAGE: Qt version 5.15.2
Project MESSAGE: Linux build
Project MESSAGE: Daily Build
Project MESSAGE: GIT_DESCRIBE v4.2.3
Project MESSAGE: QGroundControl APP_VERSION_STR VERSION v4.2.3 4.2.3
Project MESSAGE: Release flavor
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Including support for AirMap
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Including support for video streaming
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!

```

```
$> make -j8
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp: In member function ‘void QGCCacheWorker::_lookupReady(QHostInfo)’:
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp:1187:27: error: expected type-specifier before ‘QTcpSocket’
 1187 |         auto socket = new QTcpSocket();
      |                           ^~~~~~~~~~
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp:1192:26: error: ‘QTcpSocket’ has not been declared
 1192 |         connect(socket, &QTcpSocket::connected, this, [this, socket] {
      |                          ^~~~~~~~~~
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp: In lambda function:
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp:1195:13: error: ‘socket’ is not captured
 1195 |             socket->deleteLater();
      |             ^~~~~~
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp:1192:68: note: the lambda has no capture-default
 1192 |         connect(socket, &QTcpSocket::connected, this, [this, socket] {
      |                                                                    ^
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp:1187:14: note: ‘<typeprefixerror>socket’ declared here
 1187 |         auto socket = new QTcpSocket();
      |              ^~~~~~
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp: In lambda function:
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp:1200:13: error: ‘socket’ is not captured
 1200 |             socket->deleteLater();
      |             ^~~~~~
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp:1197:77: note: the lambda has no capture-default
 1197 |         connect(socket, &QAbstractSocket::errorOccurred, this, [this, socket](QAbstractSocket::SocketError error) {
      |                                                                             ^
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/QGCTileCacheWorker.cpp:1187:14: note: ‘<typeprefixerror>socket’ declared here
 1187 |         auto socket = new QTcpSocket();
      |              ^~~~~~

```
Any clue if I have missed anything or is this a bug?

ashishmagar600, 2022-07-01:
I needed to add the QTcpSocket header in QGCTileCacheWorker.cpp

`#include <QTcpSocket>`

It is past that error now. However crashes with new error:

```
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/MapProvider.cpp: In constructor ‘MapProvider::MapProvider(const QString&, const QString&, quint32, QGeoMapType::MapStyle, QObject*)’:
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/MapProvider.cpp:23:40: error: incomplete type ‘QLocale’ used in nested name specifier
   23 |     const QStringList langs = QLocale::system().uiLanguages();
      |                                        ^~~~~~
compiling /media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/EsriMapProvider.cpp
Makefile:137039: recipe for target 'MapProvider.o' failed
make: *** [MapProvider.o] Error 1
make: *** Waiting for unfinished jobs....
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/GoogleMapProvider.cpp: In member function ‘void GoogleMapProvider::_tryCorrectGoogleVersions(QNetworkAccessManager*)’:
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/GoogleMapProvider.cpp:108:27: error: variable ‘QSslConfiguration conf’ has initializer but incomplete type
  108 |         QSslConfiguration conf = qheader.sslConfiguration();
      |                           ^~~~
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/GoogleMapProvider.cpp:108:42: error: ‘class QNetworkRequest’ has no member named ‘sslConfiguration’; did you mean ‘http2Configuration’?
  108 |         QSslConfiguration conf = qheader.sslConfiguration();
      |                                          ^~~~~~~~~~~~~~~~
      |                                          http2Configuration
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/GoogleMapProvider.cpp:109:32: error: ‘QSslSocket’ has not been declared
  109 |         conf.setPeerVerifyMode(QSslSocket::VerifyNone);
      |                                ^~~~~~~~~~
/media/ashish/Data1/_ASHISH/15_QGroundControl/qgroundcontrol_source_build/qgroundcontrol_stable4p2/qgroundcontrol/src/QtLocationPlugin/GoogleMapProvider.cpp:110:17: error: ‘class QNetworkRequest’ has no member named ‘setSslConfiguration’; did you mean ‘http2Configuration’?
  110 |         qheader.setSslConfiguration(conf);
      |                 ^~~~~~~~~~~~~~~~~~~
      |                 http2Configuration
Makefile:137340: recipe for target 'GoogleMapProvider.o' failed
make: *** [GoogleMapProvider.o] Error 1

```

I am using the exactly same Qt specified in the official documentation that is Qt 5.15.2. I am wondering why the same version has such issues

Fangwq, 2022-07-29:
I have the same problem, and it's very confusing, -_-!

closed 2024-05-31 by HTRamsey