mavlink/qgroundcontrol #10350: 4.2.3: AppImage dose not work: error while loading shared libraries: libpulse-mainloop-glib.so.0

Arccoma, 2022-07-19:
[image]
<figure1.  PX4 on SITL  simulatoin with Gazebo without QGC >

I am currently trying to simulate PX4 using Gazebo.
When "command takeoff" was entered without a connection to the QGC, the FailSafe was activated and the drone's prop stopped immediately and failed to take off.

I downloaded the QGC daily version for Linux (v4.2.3) from the WSL2 environment following the instructions in the link below to connect the QGC to Gazebo to avoid FailSafe and succeed in Takeoff.
https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html

However, when running QGroundContrl.AppImage, I encountered the following error.
[image]
< Figure 2.  Fail to run QGroundContrl.AppImage >

> /tmp/.mount_QGrounXXXXXX/QGroundControl: error while loading shared libraries: libpulse-mainloop-glib.so.0: cannot open shared object file: No such file or directory
(  'XXXXXX' keeps changing every time )

I'd appreciate your help.

Arccoma, 2022-07-19:
https://github.com/mavlink/qgroundcontrol/issues/10306
After seeing this help and doing 

> sudo apt install -y libpulse-dev

I was able to run QGC. 
But the QGC screen comes out like this.
[image]

Arccoma, 2022-07-21:
[image]

Besides the computer(Ryzen Desktop) that breaks the QGC map, I tested it on two other laptops, where the map doesn't break. I don't know why the map is broken or not.

mnumanuyar, 2023-08-05:
this also happens in docker

1f-small-Xue-study-today, 2023-10-11:
So I want to run PX4 in docker, and I need to install QGC in my docker environment, how to solve this ?

lars-nagel, 2023-12-08:
I fixed the broken map issue by installing the latest onboard GPU drivers for my AMD CPU.

closed 2024-05-31 by HTRamsey