mavlink/qgroundcontrol #10360: What is wrong with Android build pipeline on github?

zdanek, 2022-08-09:
Hi.
Any of you guys, maybe @DonLakeFlyer or other maintainter could check and tell what is wrong with build pipeline?
I don't know if my PR is wrong or the pipeline is broken and android code does not compile?

Thank you.
Zdanek

DonLakeFlyer, 2022-08-10:
I'm not sure the android builds are working now. I've given up on trying to keep it working after fighting with it forever.

zdanek, 2022-08-10:
Sorry to hear that. I tried to build Android on my Ubuntu box but it no longer works. I'm not an Android guy. I just set it up a few months ago with the docs and then it worked. Now it does not build but it might be my env or latest sources problem.

I also tried to build Android QGC with a container but there's very old Docker image available that no  longer works. Does anybody know about working solution to build an Android app with a Docker? There's Ubuntu container version but it does not apply.

booo, 2022-08-22:
The android build in the CI fails after calling calling androiddeployqt after some minor other changes. You can take a look at my current attempts here: https://github.com/booo/qgroundcontrol/runs/7955471673?check_suite_focus=true#step:13:1786

I think the current problems I see are due to llvm updates: https://bugreports.qt.io/browse/QTBUG-104580

AndreySkyFoxSidorov, 2022-08-23:
Assembled very easy, Android/Windows...
What to check:
1. Android SDK(Lasttest) and NDK(21.3.6528147)
2. Install gstreamer-1.0-android-universal-1.18.5
3. Qt5.15.2 use qt-enterprise-windows-x86-5.15.2
4. You need to know the basics of building for Android...

I used this instruction: https://dev.qgroundcontrol.com/master/en/getting_started/index.html
Only thing that doesn't work is signing the build. But it can be done manually: 
`SET PATH=%PATH%;C:\AndroidSDK2\jdk-11\bin
SET PATH=%PATH%;C:\AndroidSDK2\SDK\build-tools\31.0.0\

copy /Y c:\qgraund\build-Android-Release\android-build\build\outputs\apk\debug\android-build-debug.apk android-build-debug.apk 
jarsigner -verbose -keystore android_release.keystore -storepass ******* android-build-debug.apk qgc
del qgroundcontrol-signed.apk
rename android-build-debug.apk qgroundcontrol-signed.apk
`

booo, 2022-08-24:
@AndreySkyFoxSidorov automate the process in a script that we can deploy as github action?

fredowski, 2022-08-26:
> I think the current problems I see are due to llvm updates: https://bugreports.qt.io/browse/QTBUG-104580

I think the problem is indeed a changed llvm version. The bugfix is already in QT: https://codereview.qt-project.org/c/qt/qtbase/+/418757/4/src/tools/androiddeployqt/main.cpp

The problem is that the QT 5.15 version is only maintained for commercial users by QT and the open source version will only be updated after a year or so? KDE maintains a QT5 version by applying selected patches and the patch also landed there:

https://invent.kde.org/qt/qt/qtbase/-/commit/40ddf2c7f34e429af4d18063ca947490cdc12ba9

As far as I understand it to follow that path would mean to build QT5 via KDE from scratch or wait for a new 5.15 release from QT which includes that fix.

As a workaround I switched back the NDK version to r23c and the build goes a bit further. The build stops probably because my NDK version switch was only done partly. My work up to now is here:

https://github.com/fredowski/qgroundcontrol/pull/2

zdanek, 2022-08-26:
@fredowski how can it be a problem with QT as QGC uses fixed 5.15.2 version so such problems does not emerge?
Am I missing something?

An error that you have and I have it as well is 
>   -- Skipping /home/runner/work/_temp/Qt/5.15.2/android/plugins/position/libplugins_position_qtposition_serialnmea_armeabi-v7a.so. It has unmet dependencies: lib/libQt5SerialPort_armeabi-v7a.so.

which seems some problems with dependency library for nmea. 
This happens when run with `make apk`

fredowski, 2022-08-26:
@zdanek if you look at the last CI run for Android 32bit for commit ["libevents: update submodule"](https://github.com/mavlink/qgroundcontrol/runs/8031541083?check_suite_focus=true) you will find the following error message in line 4241:

`/usr/local/lib/android/sdk/ndk/25.0.8775105/toolchains/llvm/prebuilt/linux-x86_64/bin/llvm-readobj: error: unknown argument '--libs'`

This indicates that ndk version 25.0.8775105 was used for the build. That ndk version [was released on 13.7.2022](https://groups.google.com/g/android-ndk-announce/c/P0hK8_G5Sms), i.e. it is more or less the latest ndk version. So it seems that a fixed Qt version 5.15.2 does not mean that the underlying ndk version is fixed.

fredowski, 2022-08-27:
The github ubuntu-20.04 build machine provides the android sdk and ndk. See:

https://github.com/actions/runner-images/blob/ubuntu20/20220821.1/images/linux/Ubuntu2004-Readme.md

fredowski, 2022-08-28:
I do not have an Android tablet. Can maybe somebody test the resulting apks from the pr?

booo, 2022-08-28:
Apk works on my mobile. 

I'm closing this as done.

closed 2022-08-28 by booo

zdanek, 2022-08-30:
I've also checked. Works great.

fredowski, 2022-08-30:
@zdanek @booo : Thanks for checking the PR #10380.

commit 0b19305c (2022-09-06): fix Android CI for 32bit and 64bit builds

  * install and use NDK r21e
  * update QT install action from v2 to v3

Starting with NDK Version r24 llvm will not accept one-dash
long options anymore. This is bug QTBUG-104580

https://bugreports.qt.io/browse/QTBUG-104580

The r23c NDK version did not work due to issues with
stripping symbols. So I install and use r21e. QT 5.15.2 uses
gradle version 5.6.4 with the default android gradle plugin 3.6.0.
With gradle 6.9.2 and android gradle plugin 4.2.2 the 32bits
build worked. NDK r22b is not compatible with gstreamer 1.18 and
results in a failing 64 [... cut]