mavlink/qgroundcontrol #10366: RTSP video not recording in Android

AndreySkyFoxSidorov, 2022-08-12:
## Expected Behavior
RTSP video recording not working

## Current Behavior
[D] at qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:103 - "toggleShooting true"
[!] at ../../qgroundcontrol/qgroundcontrol/src/VideoReceiver/GstVideoReceiver.cc:531 - "Failed to link valve and file sink "rtsp://192.168.144.25:8554/ch01.264""

## Steps to Reproduce:
1. Build Android ( Commit: 2f14b59793167e885b50a71bcfe1a8c8cfcff071 [2f14b59] ) or Install Release version v4.2.3
2. Run qgroundcontrol
3. Set rtsp settings. 
4. Open camera. 
5. I can see the video, but an error occurs.
[D] at qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:103 - "toggleShooting true"
[!] at ../../qgroundcontrol/qgroundcontrol/src/VideoReceiver/GstVideoReceiver.cc:531 - "Failed to link valve and file sink "rtsp://192.168.144.25:8554/ch01.264""

## System Information
When posting bug reports, include the following information
- Operating System: [Android 9, FPV SIYI MK15 ]
- Flight Controller: [Pixhawk]
- Qgroundcontrol: version v4.2.3

## Detailed Description
I think the problem is the wrong choice of video codec h265. Perhaps you need to add a field with the choice of codec to the settings for rtsp

## Log Files and Screenshots
[D] at qrc:/qml/QGroundControl/FlightMap/PhotoVideoControl.qml:103 - "toggleShooting true"
[!] at ../../qgroundcontrol/qgroundcontrol/src/VideoReceiver/GstVideoReceiver.cc:531 - "Failed to link valve and file sink "rtsp://192.168.144.25:8554/ch01.264""

jensensi, 2022-08-19:
Hello,

I also experience the same behavior on my Android device, when following the same steps.
Video is displayed on the device, but the recording fails.

- I receive the video stream over RTSP.
- I have selected mkv as the recording format.

When I build to Windows and run on a PC, the recording is initiated perfect.
When I build to Android and run on my device, the software is unable to link _recorderValve and _fileSink in GstVideoReceiver.cc

Module:  GstVideoReceiver.cc
Function:  void GstVideoReceiver::startRecording(const QString& videoFile, FILE_FORMAT format)
Call:  gst_element_link(_recorderValve, _fileSink) return FALSE

**System Information**

- Operating System: [Android 9, FPV SIYI MK15 ]
- Qgroundcontrol: version v4.2.3
- Gstreamer: 1.18.1 (on both Windows and Android, same behavior with 1.18.5 on Android)

Codec information when streaming to VLC (on Android and Windows).

- Codec: MPEG-H Part2/HEVC (H.265) (hevc)
- Type: Video
- Viceo resolution: 2560x 1440
- Frame rate: 29.970030

**Detailed Description**
I have investigated the template and pad information in _recorderValve and _fileSink using qCDebug traces based on
"Basic tutorial 6: Media formats and Pad Capabilities"
https://gstreamer.freedesktop.org/documentation/tutorials/basic/media-formats-and-pad-capabilities.html?gi-language=c

**Findings: (so far)**
The 'src pad' capabilities change from Windows to Android.

      video/x-h265
              alignment: au
          stream-format: hvc1		->  changed from 'hvc1' in Windows to 'byte-stream' in Android (?)
                 parsed: true
                  width: 2560
                 height: 1440
          chroma-format: 4:2:0
         bit-depth-luma: 8
       bit-depth-chroma: 8
            colorimetry: 1:3:5:1
                profile: main
                   tier: main
                  level: 5.1

_Questions: (to investigate further)_
Is this a valid reason why _recorderValve and _fileSink is unable to link?
Why do the 'src pad' capabilities change from Windows to Android when the stream is unchanged?

**Log Files and Screenshots**

_Windows: (success)_

```
Starting recording "rtsp://192.168.0.130:554/1"
New video file: "C:/Users/Privat/Documents/QGroundControl Daily/Video/2022-08-18_16.35.59.mkv"  "rtsp://192.168.0.130:554/1"

Pad Templates for RTSP packet receiver:
  SRC template: 'stream_%u'
    Availability: Sometimes
    Capabilities:
      application/x-rtp
      application/x-rdt

Pad Templates for File Sink:
[... 176 lines cut ...]
                 height: [ 16, 2147483647 ]
      video/x-vp8
                  width: [ 16, 2147483647 ]
                 height: [ 16, 2147483647 ]
      video/x-vp9
                  width: [ 16, 2147483647 ]
                 height: [ 16, 2147483647 ]
      video/x-raw
                 format: { (string)YUY2, (string)I420, (string)YV12, (string)UYVY, (string)AYUV, (string)GRAY8, (string)BGR, (string)RGB }
                  width: [ 16, 2147483647 ]
                 height: [ 16, 2147483647 ]
      video/x-prores
                  width: [ 16, 2147483647 ]
                 height: [ 16, 2147483647 ]
      video/x-wmv
             wmvversion: [ 1, 3 ]
                  width: [ 16, 2147483647 ]
                 height: [ 16, 2147483647 ]
      video/x-av1
                  width: [ 16, 2147483647 ]
                 height: [ 16, 2147483647 ]

Failed to link valve and file sink "rtsp://192.168.0.130:554/1"

```

AndreySkyFoxSidorov, 2022-08-23:
As I understand it, the reason is that _recorderValve _fileSink does not connect. 

Class GstVideoReceiver
Metod GstVideoReceiver::startRecording(const QString& videoFile, FILE_FORMAT format)
Code: 
`    if (!gst_element_link(_recorderValve, _fileSink)) {
        qCCritical(VideoReceiverLog) << "Failed to link valve and file sink" ;
        _dispatchSignal([this](){
            emit onStartRecordingComplete(STATUS_FAIL);
        });
        return;
    }`

I have a suspicion that the recorder cannot get a keyframe to start working on the updated pipeline...
Any ideas how to fix this?

AndreySkyFoxSidorov, 2022-08-23:
FIX!
Problem with asynchronous link
[image]
Need Replase:
`
    if (!gst_element_link(_recorderValve, _fileSink)) {
        qCCritical(VideoReceiverLog) << "Failed to link valve and file sink" << _uri << " format:" <<format << " sink:" <<  _fileSink;
        _dispatchSignal([this](){
            emit onStartRecordingComplete(STATUS_FAIL);
        });
        return;

`
To:
`
    int ErrorCount = 50000;
    while (!gst_element_link(_recorderValve, _fileSink))
    {
        qCCritical(VideoReceiverLog) << "ASYNK gst_element_link ERROR";
        ErrorCount--;
        if(ErrorCount <=0)
        {
            qCCritical(VideoReceiverLog) << "Failed to link valve and file sink" << _uri << " format:" <<format << " sink:" <<  _fileSink;
            _dispatchSignal([this](){
                emit onStartRecordingComplete(STATUS_FAIL);
            });
            return;
        }
    }
`

andrewvoznytsa, 2022-08-23:
I didn't read whole thread but the fix you are proposing is not acceptable. There is no any good reason to have such loop. You need to follow gstreamer design and continue linking when elements are ready for that, instead of wasting CPU cycles and blocking thread.

Next time please prepare PR and work in a github way.

andrewvoznytsa, 2022-08-23:
Also, your assumption about the key frame is wrong. It is filtered [there](https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/GstVideoReceiver.cc#L555) so file sink always receives the key frame first.

AndreySkyFoxSidorov, 2022-08-23:
> Also, your assumption about the key frame is wrong. It is filtered [there](https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/GstVideoReceiver.cc#L555) so file sink always receives the key frame first.

I bought a mk15 camera with MPEG-H Part2/HEVC (H.265) (hevc) codec. The problem is not only with me but also with other users:
https://www.facebook.com/groups/siyitech/posts/1183273419179537/ (for example)
This problem is for all users who use such codecs. A more elegant solution to this problem would be ideal. I tried different versions of gstreamer from 1.18.5 to 1.20.3, the behavior is the same.
Searching the Internet, I came across the same problems in other applications (based on gstreamer), they are fixed by making delays or stopping and restarting the video stream with recording.
I don't know how to solve this problem elegant ...

andrewvoznytsa, 2022-08-24:
This problem is most likely related to the sink pad, which is not exposed at the time of linking. There are at least one similar case in the video receiver with appropriate handling.

First of all you need to identify your problem. There is perfect [starting pont](https://gstreamer.freedesktop.org/documentation/tutorials/basic/debugging-tools.html?gi-language=c) for that. Also I'd suggest to continue in a way used by @jensensi in his [post](https://github.com/mavlink/qgroundcontrol/issues/10366#issuecomment-1220475094) - it is much easier and clear in this way for everyone and saves a lot of time. This is super important for guys who are not paid for this and volunteer on their own.

hmloww, 2022-08-28:
I've encountered the same issue with 4.2.3 on Android 12.
Although the record video button does not seem to function, an empty .mkv (i selected .mkv) is created everytime the record button is pressed.

fredowski, 2022-09-07:
I tried the Android build results of PR #10396 on a Samsung Galaxy M20 running Android 10 and with that change I can display and record the following rtsp stream: 

`rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mp4`

The APKs from that pull request are here: https://github.com/mavlink/qgroundcontrol/actions/runs/3008425702

#10396 basically reverts the switch from decodebin to decodebin3 and reintroduces the specialhandlings from @andrewvoznytsa . See also #9838 . Can you maybe try the APKs on your devices and check if the display and recording works? I tried the following streams with videodecoder set to default and 

I set the stream source in QGC to "UDP h.264 Video Stream" and try

`gst-launch-1.0 -vvv videotestsrc ! "video/x-raw,width=320,height=240,format=(string)Y444" ! x264enc ! h264parse ! rtph264pay! udpsink host=192.168.1.101 port=5600`

`gst-launch-1.0 -vvv videotestsrc ! "video/x-raw,width=320,height=240,format=(string)I420" ! x264enc ! h264parse ! rtph264pay! udpsink host=192.168.1.101 port=5600` 

Setting the QGC Video Source to "RTSP Video Stream" with "RTSP URL" set to 

`rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mp4`

AndreySkyFoxSidorov, 2022-09-08:
YA nashel pochemu nezapisyvayet'sya video. V nekotorykh kamerakh otsutstvuyet DONL pole. I etu problemu ne sovsem ponyatno kak pofiksit'.
129 / 5 000
Результаты перевода
I found why the video is not being recorded.
Some cameras do not have a DONL field.
And this problem is not entirely clear how to fix it.

LOG:
[I] at ../gst/rtp/gstrtph265depay.c:1284 - "<rtph265depay3> Assuming DONL field is not present"

jensensi, 2022-10-12:
Hello again,

I have tried the APK provided by fredowski ([above](https://github.com/mavlink/qgroundcontrol/issues/10366#issuecomment-1239570348)) on my device to check if the display and recording works?
I am mainly interested in RTSP, but I have tried all three suggested streams with videodecoder set to default.

In this post I provide the results of the test with the APK. 

**System Information**
Operating System: [Android 9, FPV SIYI MK15 ]
Qgroundcontrol: [QGroundControl64.apk](https://github.com/mavlink/qgroundcontrol/actions/runs/3008425702) provided by fredowski

**Test**
_1) Source: "UDP h.264 Video Stream". File format: "mkv" (using gst-launch-1.0 iwth Y444 & I420) Video decode priority: "Default"._

Stream is started. QGC is launched. 
- The video display the video stream. The recording video button is enabled (red/round).
When the record button is pressed, the button change shape from round to square (recording started) and remains square until it is pressed again (recording stopped).
- The video folder contains new video recording (for Y444 and I420). The Y444 format video CANNOT play. The I420 format video CAN play.

_2) Source: "RTSP Video Stream". File format: "mkv" (url = Bunny). Video decode priority: "Default"._

Stream is started. QGC is launched. 

One of two results are experienced when application is launched:

- The video display "WAITING FOR VIDEO". No video is displayed. The recording video button is enabled (red/round).
Nothing happens when the record button is pressed, i.e. the button is unchanged (red/round).

- The video display some flickering gui elements (as described in #10396). The recording video button is enabled (red/round).
When the record button is pressed, the button change shape from round to square (recording started) and remains square until it is pressed again (recording stopped).

In this latter case, the video folder contain a new video recording, which is a recording of the 'Bunny' video stream (no flickering gui elements here).

**Log files**
I have included some log files for the two possible outcomes of test 2) (above) using the following settings:
- GStreamerAPILog enabled
- GStreamerLog ensbled
- VideoReceiverLog ensbled
- GStreamer Debug = Error

_Result: WAITING FOR VIDEO_

[D] at ../../qgroundcontrol/qgroundcontrol/src/QGCApplication.cc:281 - "Settings location "/data/user/0/org.mavlink.qgroundcontrolbeta/files/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true"
[D] at ../../qgroundcontrol/qgroundcontrol/src/QGCLoggingCategory.cc:120 - "Filter rules "*Log.debug=false\nGStreamerLog.debug=true\nGStreamerAPILog.debug=true\nVideoReceiverLog.debug=true\nGStreamerAPILog.debug=true\nqt.qml.connections=false""
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:551 - "(NULL) Unknown color format 0x7fa30c06, ignoring"
[!] at ../sys/androidmedia/gstamc.c:2006 - "(NULL) Unable to map AAC profile 0x00000005"
[!] at ../sys/androidmedia/gstamc.c:2006 - "(NULL) Unable to map AAC profile 0x0000001d"
[!] at ../sys/androidmedia/gstamc.c:2006 - "(NULL) Unable to map AAC profile 0x00000017"
[!] at ../sys/androidmedia/gstamc.c:2006 - "(NULL) Unable to map AAC profile 0x00000027"
[!] at ../sys/androidmedia/gstamc.c:2006 - "(NULL) Unable to map AAC profile 0x00000014"
[!] at ../sys/androidmedia/gstamc.c:2006 - "(NULL) Unable to map AAC profile 0x0000002a"
[!] at ../sys/androidmedia/gstamc.c:2254 - "(NULL) Unable to map H264 profile 0x00080000"
[!] at ../sys/androidmedia/gstamc.c:2254 - "(NULL) Unable to map H264 profile 0x00010000"
[!] at ../sys/androidmedia/gstamc.c:2254 - "(NULL) Unable to map H264 profile 0x00080000"
[!] at ../sys/androidmedia/gstamc.c:2254 - "(NULL) Unable to map H264 profile 0x00010000"
[!] at ../sys/androidmedia/gstamc.c:2330 - "(NULL) Unable to map H265 profile 0x00000004"
[!] at ../sys/androidmedia/gstamc.c:2254 - "(NULL) Unable to map H264 profile 0x00080000"
[!] at ../sys/androidmedia/gstamc.c:2254 - "(NULL) Unable to map H264 profile 0x00010000"
[!] at ../sys/androidmedia/gstamc.c:2254 - "(NULL) Unable to map H264 profile 0x00010000"
[... 80 lines cut ...]
[E] at qrc:/toolbar/GPSRTKIndicator.qml:93 - "qrc:/toolbar/GPSRTKIndicator.qml:93: TypeError: Cannot read property 'active' of null"
[E] at qrc:/toolbar/GPSRTKIndicator.qml:105 - "qrc:/toolbar/GPSRTKIndicator.qml:105: TypeError: Cannot read property 'numSatellites' of null"
[... cut ...]
[E] at ../../qgroundcontrol/qgroundcontrol/src/VideoManager/SubtitleWriter.cc:111 - "Attempting to capture fact data with no active vehicle!"
[E] at ../../qgroundcontrol/qgroundcontrol/src/VideoManager/SubtitleWriter.cc:111 - "Attempting to capture fact data with no active vehicle!"
[E] at ../../qgroundcontrol/qgroundcontrol/src/VideoManager/SubtitleWriter.cc:111 - "Attempting to capture fact data with no active vehicle!"
[E] at ../../qgroundcontrol/qgroundcontrol/src/VideoManager/SubtitleWriter.cc:111 - "Attempting to capture fact data with no active vehicle!"

fredowski, 2022-10-12:
Thanks for testing on Android and writing the detailed test report! Unfortunately you have tested a version from the Pull Request #10396 which now only addresses some problems on MacOS. I have moved out all non MacOS parts to a separate PR #10420.  The flickering problem is addressed in #10420. The RTSP problem that the video is sometimes not shown as you also noticed was fixed in #10409 which is merged already. I rebased #10420 to include the RTSP fix.

Can you try with the Android builds https://github.com/mavlink/qgroundcontrol/actions/runs/3237541415 from PR #10420? 

Maybe you can report your test results then in PR #10420?

jensensi, 2022-10-13:
Wow, I can tell that you have been busy, while I went away. Thanks!

I will test the new Android builds from PR [#10420](https://github.com/mavlink/qgroundcontrol/pull/10420) and then report my findings also to PR [#10420](https://github.com/mavlink/qgroundcontrol/pull/10420)

HTRamsey, 2024-05-31:
This was fixed recently, (Again?)

closed 2024-05-31 by HTRamsey