mavlink/qgroundcontrol #10368: The ordination in the QGC combobox for camera selection is inconsistent

joaoantoniocardoso, 2022-08-12:
Originally from https://github.com/bluerobotics/BlueOS-docker/issues/1011:

## Expected Behavior
The order should be consistent, possibly alphabetically sorted.

## Current Behavior
Every time I restart QGC the order of the cameras is shown differently.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Setup multiple MAVLink cameras (I personally recommend [mavlink-camera-manager t3.6.2](https://github.com/bluerobotics/mavlink-camera-manager/releases/tag/t3.6.2))
2. Open QGC, take note of the order on the combobox camera selection
3. Restart QGC many times, the order will be different.

## System Information
When posting bug reports, include the following information
- Operating System: Arch Linux
- QGC Version: tested with AppImage [v4.2.3](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.2.3), but should happen with any other too.
- QGC build: AppImage downloaded from [here](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.2.3) 
- Flight Controller: Pixhawk 1
- Autopilot (with version): ArduPilot (Submarine) 4.2.0 (DEV)

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
1st open: [image]
4th reopen: [image]

Thanks!

zdanek, 2022-08-24:
I can fix it after my PR #10349 will go into master. I made some fixes and tidy ups so fixing this issue is not a problem for me.
But I want to wait for my PR to be merged.

HTRamsey, 2024-05-31:
Please reopen if zdaneks additions did not fix this

closed 2024-05-31 by HTRamsey