mavlink/qgroundcontrol #10378: After Commit: 3f479fc Video is not working

AndreySkyFoxSidorov, 2022-08-23:
After Commit: 3f479fc Video is not working. Platform Windows/Android

Log:
(QGroundControl.exe:20868): GLib-GObject-WARNING **: 15:45:44.989: invalid uninstantiatable type '<unknown>' in cast to 'GstBin'
(QGroundControl.exe:20868): GStreamer-CRITICAL **: 15:45:44.994: gst_bin_add_many: assertion 'GST_IS_BIN (bin)' failed
(QGroundControl.exe:20868): GLib-GObject-WARNING **: 15:45:44.998: invalid uninstantiatable type '<unknown>' in cast to 'GstElement'
(QGroundControl.exe:20868): GStreamer-CRITICAL **: 15:45:44.999: gst_element_add_pad: assertion 'GST_IS_ELEMENT (element)' failed
(QGroundControl.exe:20868): GLib-GObject-CRITICAL **: 15:45:44.999: g_object_set: assertion 'G_IS_OBJECT (object)' failed
VideoReceiverLog:
(QGroundControl.exe:20868): GLib-GObject-WARNING **: 15:45:45.001: invalid uninstantiatable type 'void' in cast to 'GstBin'
 Unable to find sin
(QGroundControl.exe:20868): GStreamer-CRITICAL **: 15:45:45.002: gst_bin_add_many: assertion 'GST_IS_BIN (bin)' failed
k pad of video sink "rtsp://192.168.0.40:8554/main.265"
(QGroundControl.exe:20868): GLib-GObject-WARNING **: 15:45:45.007: invalid uninstantiatable type 'void' in cast to 'GstElement'
(QGroundControl.exe:20868): GStreamer-CRITICAL **: 15:45:45.012: gst_element_add_pad: assertion 'GST_IS_ELEMENT (element)' failed
(QGroundControl.exe:20868): GLib-GObject-CRITICAL **: 15:45:45.013: g_object_set: assertion 'G_IS_OBJECT (object)' failed

fredowski, 2022-09-02:
Does that mean that Video is not displayed in the QGC app? I try to fix the Android CI in PR https://github.com/mavlink/qgroundcontrol/pull/10380#issuecomment-1234784608 but the Android QGC app does not display the video stream.

fredowski, 2022-09-06:
This is fixed with 9b63448fc0631.

closed 2022-09-08 by AndreySkyFoxSidorov