mavlink/qgroundcontrol #10382: Parameter Editor's text format is weird

junwoo091400, 2022-08-29:
## Expected Behavior
Texts shouldn't overlap with each other and be displayed in a correct way

## Current Behavior
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.Flash PX4 `main`
2. Check EKF2_HGT_REF parameter's descriptions

booo, 2022-11-06:
Did this get fixed? Skimming through the referenced issue I can't really tell.

junwoo091400, 2022-11-09:
> Did this get fixed? Skimming through the referenced issue I can't really tell.

Nope, not yet 😢, do you have a clue on how to fix it? I noticed this today again and reminded me how annoying this is haha

HTRamsey, 2024-05-31:
Seems to be good now, please reopen if not.

closed 2024-05-31 by HTRamsey