mavlink/qgroundcontrol #10385: Unable to add waypoints for ground rover vehicles

Jaeyoung-Lim, 2022-08-31:
## Expected Behavior
Be able to plan a  mission by adding waypoints

## Current Behavior
When trying to add a waypoint to the mission, it is not possible to add a waypoint for a flight plan.

## Steps to Reproduce:
1. Connect QGC to a ground rover vehicle
2. Enter plan view
3. Not possible to add waypoints

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: 4.2.1
- QGC build: stable
- Flight Controller: SITL Gazebo
- Autopilot (with version): PX4 `v1.13.0` and `main`

## Detailed Description
It seems that for ground rover vehicles it is not possible to add waypoints and therefore not possible to plan a mission (at all)

## Log Files and Screenshots
[image]

mvp-cwlin, 2022-08-31:
In the Application Setting, try to enable "Mission Do Not Require Takeoff Item"

[image]

Jaeyoung-Lim, 2022-09-01:
@mvp-cwlin Thanks, but shouldn't this not be necessary, since rovers are able to handle takeoff waypoints?

mvp-cwlin, 2022-09-01:
If you change the vehicle type setting on QGroundControl to Multi-Rotor, it will let you add a takeoff item, then you can add waypoints. And yes, I believe rovers can handle takeoff.

Jaeyoung-Lim, 2022-09-01:
@mvp-cwlin I am not sure if I agree

QGC already knows which vehicle type it is so it should not prevent the user from planning a mission

mvp-cwlin, 2022-09-01:
I agree, QGC should automatically configure to not require a takeoff item when it knows it's a rover.

closed 2026-10-05 by github-actions[bot]