mavlink/qgroundcontrol #10388: Inconsistent video stream list after restarting mavlink video streams.

joaoantoniocardoso, 2022-09-01:
## Expected Behavior
Multiple video streams should show consistently despite mavlink instabilities.

## Current Behavior
When this quick reset happens, QGroundcontrol starts to show inconsistent behavior on how it deals with multiple streams. See the video below, pay attention to the multiple cameras shown, and how, after each reset, it becomes worst:
- completely looses track one of the streams.
- duplicates one of the streams, but the wrong controls, which probably means the wrong component ID.

I didn't show in the video, but if I restart the QGroundControl, it medially finds all the streams again, although with a different order, as already reported in #10368.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Setup a multiple-camera stream (I'm using mavlink-camera-manger)
2. For a small period of time, cut the mavlink connection of the cameras - I am using the "reset" functionality of mavlink-camera-manager to simulate this quick freezing in the camera's heartbeat and stream, literally as if we were having some connectivity problem between QGC and the cameras in a multiple-camera setup.
3. Observe the stream selection combobox and the controls for each stream.

## System Information
When posting bug reports, include the following information
Operating System: Arch Linux
    QGC Version: tested with AppImage [v4.2.3](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.2.3), but should happen with any other too.
    QGC build: AppImage downloaded from [here](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.2.3)
    Flight Controller: Pixhawk 1 or SITL
    Autopilot (with version): ArduPilot (Submarine) 4.2.0 (DEV)

## Log Files and Screenshots

[QGC_inconsistent_example_log.txt](https://github.com/mavlink/qgroundcontrol/files/9473017/QGC_inconsistent_example_log.txt)

https://user-images.githubusercontent.com/5920286/188002548-6f1e63b0-4354-4cb4-854c-125431ac59cf.mp4

zdanek, 2022-09-02:
Great insight. I say that video code in QGC is a bit  messy. I made a first step to fix it in #10349 

There are some things to fix - like not working gstreamer streams under Android (and Windows?) and other minor issues. Then this can be fixed as well but not all at the same time as it would be difficult to track all those changes.
I can fix your problems as soon my pending video changes will jump into the repo.

HTRamsey, 2024-05-31:
Please reopen if zdaneks additions did not fix this

closed 2024-05-31 by HTRamsey