mavlink/qgroundcontrol #10395: Cannot build an android app for Target sdk version 31

praveengi113, 2022-09-06:
I've been trying to build the Android app with Target sdk version of 31. But due to some reasons, in QT creator it's only gives the output app with the target sdk version of 28. So I've tried to override the sdk build tools in gradle and I've built the app for my requirement.

But it's not opening.
As per some discussions in stack overflow and others, it's been told that the sdk build tools version more than 30 don't have 2 files(dx.jar and dx.bat) in the build tools folder. So it's been recommended to rename d8.jar to dx.jar and d8.bat to dx.bat then built the app. But no luck.

As per my observations, this is due to the gradle plugin version.

Can you give any solution for this.

booo, 2022-09-13:
I think we need a bit more information. Can you post logs from the problem?

guruz, 2022-12-08:
@praveengi113 Did you find a solution?

kjersbry, 2024-05-08:
This is also a problem in the github actions workflow for 4.4.0 and 4.3.0 (not sure about earlier versions). Meaning that, these apks for example
https://github.com/mavlink/qgroundcontrol/actions/runs/8813775690
https://github.com/mavlink/qgroundcontrol/releases/tag/v4.3.0
have targetSdkVersion=28, not 29 or 30 like the manifest file says.
This makes Google Play Protect complain with "Unsafe app blocked - This app was built for an older version of android [...]" if you try to install them.

In my 4.3.0-based fork, the solution was to add the following to qgroundcontrol.pro
```
AndroidBuild {
    ANDROID_TARGET_SDK_VERSION = 29
}
```
as suggested [here](https://forum.qt.io/post/631881).

It looks like this has been solved differently in master, after the transition to Qt6. Qt 6.1 introduces CMake target property QT_ANDROID_TARGET_SDK_VERSION, which is used [here](https://github.com/mavlink/qgroundcontrol/blob/35f8864889d7157f772708f8aad41a269c834cd2/CMakeLists.txt#L294). So I believe the above fix should not be necessary in master, only in older Qt5-based versions.

closed 2024-05-31 by HTRamsey