mavlink/qgroundcontrol #10429: Add PIDTuning UI for Rover type vehicles

max-itsl, 2022-09-22:
Good day!

Recently [Jaeyoung-Lim](https://github.com/Jaeyoung-Lim) made PID tuning section available for fixed wing with this PR #10384  https://github.com/mavlink/qgroundcontrol/pull/10384

Please consider adding the same functions for Rover frame type.

I need to fine tune PX4 Rover speed and steering. It's a big diesel tractor, cruise speed is below 2 m/s. I'm getting speed difference for uphill vs downhill up to 0.5 m/s, which heavily affects rover behavior in sharp turns. I need to sharpen speed controller response and remove oscillations. Now my method is: sitting inside the tractor, monitor gas pedal behavior while modifying P, I and D values. As you can guess, this method is quite inaccurate.

Previously I've done it using Ardupilot and Mission Planner as described here https://ardupilot.org/rover/docs/rover-tuning-throttle-and-speed.html#desired-speed-to-throttle-pid-tuning , I've used “pidachieved” and “piddesired” monitoring tool in Mission Planner. Now I'm trying to reproduce it using PX4.

- Q: What flight stacks must it work with? (All, PX4, ArduPilot)
- A: PX4

- Q: Relevant vehicle types?
- A: Rover

- Q: Are there other systems that have this feature?
- A: Yes, Ardupilot with Mission Planner

Thank you

Max

junwoo091400, 2022-09-22:
Hi! I will be able to work on this 😊. Btw are you on px4 slack? We could chat in the #rover channel to discuss your project as well 🤔

max-itsl, 2022-09-22:
Hi! Thank you. Yes, I'm aware of the #rover channel. I decided to publish this request here as there is that similar PR from Jaeyoung-Lim . I'm trying to make a tractor work alone in the field.

Jaeyoung-Lim, 2022-09-30:
@junwoo091400 Do you have an ETA on when you would work on this? Otherwise I can already do it in case you haven't started yet

junwoo091400, 2022-09-30:
> @junwoo091400 Do you have an ETA on when you would work on this? Otherwise I can already do it in case you haven't started yet

Got side tracked 👻, yes sure you can go ahead! I haven't made any changes yet

max-itsl, 2022-09-30:
@Jaeyoung-Lim  Thank you. I will test this feature right away as it gets available

junwoo091400, 2022-10-13:
Any updates? @Jaeyoung-Lim 

Btw just found this PR that originally implemented this feature: https://github.com/mavlink/qgroundcontrol/pull/6295

junwoo091400, 2022-11-10:
As I am now preparing to do a boat demo this week, I got a chance to work on this. The WIP PR is here: https://github.com/mavlink/qgroundcontrol/pull/10483

@max-itsl

closed 2026-10-04 by github-actions[bot]