mavlink/qgroundcontrol #10435: Last stable version: crash at startup on MacOS 12.6 (up to date)

krl91, 2022-09-27:
## Expected Behavior
qgroundControl should start...

## Current Behavior
qgroundControl (last stable version) crashes at startup all the time 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. try to open qgroundcontrol
2. it crashes (all the time)

## System Information
When posting bug reports, include the following information
- Operating System: MacOS 12.6 (up to date - Intel)
- QGC Version: current last stable version from http://qgroundcontrol.com/downloads/

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
```
Applications/QGroundControl.app/Contents/MacOS/QGroundControl
dyld[41821]: Library not loaded: '@executable_path/../Frameworks/SDL2.framework/Versions/A/SDL2'
  Referenced from: '/Applications/QGroundControl.app/Contents/MacOS/QGroundControl'
  Reason: tried: '/Applications/QGroundControl.app/Contents/MacOS/../Frameworks/SDL2.framework/Versions/A/SDL2' (no such file), '/Library/Frameworks/SDL2.framework/Versions/A/SDL2' (no such file), '/System/Library/Frameworks/SDL2.framework/Versions/A/SDL2' (no such file)
zsh: abort      /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
```

Report from MacOS
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               QGroundControl [42273]
Path:                  /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.QGroundControl
Version:               4.2.3 (4.2.3)
Code Type:             X86-64 (Native)
Parent Process:        zsh [9118]
Responsible:           iTerm2 [9110]
User ID:               501

Date/Time:             2022-09-27 21:16:08.6345 +0200
OS Version:            macOS 12.6 (21G115)
Report Version:        12
Bridge OS Version:     3.0 (14Y910)
Anonymous UUID:        360EC00B-F06D-88BB-59E0-AA19C3F608C9

Sleep/Wake UUID:       1607BA42-8ADF-4CF1-89EF-4204540C3516

Time Awake Since Boot: 130000 seconds
Time Since Wake:       2516 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: '@executable_path/../Frameworks/SDL2.framework/Versions/A/SDL2'
Referenced from: '/Applications/QGroundControl.app/Contents/MacOS/QGroundControl'
Reason: tried: '' (no such file), '' (no such file), '' (no such file)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x10c6f198e __abort_with_payload + 10
[... 139 lines cut ...]
      "treatmentId" : "5b729754-3882-4560-826f-9c0bc27d77c6",
      "experimentId" : "6192fb082171a2330e561df0",
      "deploymentId" : 400000035
    }
  ]
}
}

Model: MacBookPro13,2, BootROM 451.140.1.0.0, 2 processors, Dual-Core Intel Core i5, 2,9 GHz, 16 GB, SMC 2.37f25
Graphics: Intel Iris Graphics 550, Intel Iris Graphics 550, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x157), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB30Bus
USB Device: Apple T1 Controller
USB Device: AppleUSBUserHCI
USB Device: AppleUSBUserHCI
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5

```

fredowski, 2022-10-04:
Could you try the MacOS version from here: https://github.com/mavlink/qgroundcontrol/actions/runs/3064024101 ?

That is from my PR https://github.com/mavlink/qgroundcontrol/pull/10396 to fix the CI pipeline for MacOS.

fcrevier, 2022-10-06:
This worked for me, thanks! :)

fredowski, 2022-10-06:
thanks for checking! Could you maybe describe what you tested on which platform?

krl91, 2022-10-06:
Hi,
it does not work on my side:
`/Applications/QGroundControl.app/Contents/MacOS/QGroundControl
dyld[32703]: Library not loaded: '@executable_path/../Frameworks/SDL2.framework/Versions/A/SDL2'
  Referenced from: '/Applications/QGroundControl.app/Contents/MacOS/QGroundControl'
  Reason: tried: '/Applications/QGroundControl.app/Contents/MacOS/../Frameworks/SDL2.framework/Versions/A/SDL2' (no such file), '/Library/Frameworks/SDL2.framework/Versions/A/SDL2' (no such file), '/System/Library/Frameworks/SDL2.framework/Versions/A/SDL2' (no such file)
zsh: abort      /Applications/QGroundControl.app/Contents/MacOS/QGroundControl`

fredowski, 2022-10-06:
Very strange. Can you check via terminal inside the app if `/Applications/QGroundControl.app/Contents/MacOS/../Frameworks/SDL2.framework/Versions/A/SDL2` really does not exist? So for me it looks like this:

```
Fredo:Frameworks fritz$ cd SDL2.Framework/
Fredo:SDL2.Framework fritz$ ls
Headers		Resources	SDL2		Versions
Fredo:SDL2.Framework fritz$ cd Versions/
Fredo:Versions fritz$ ls
A	Current
Fredo:Versions fritz$ cd A/
Fredo:A fritz$ pwd
/Users/fritz/searchwing/QGroundControl-fixci.app/Contents/Frameworks/SDL2.Framework/Versions/A
Fredo:A fritz$ ls -la
total 5760
drwxr-xr-x@  6 fritz  staff      192 11 Sep 01:29 .
drwxr-xr-x@  4 fritz  staff      128 11 Sep 01:29 ..
drwxr-xr-x@ 67 fritz  staff     2144 11 Sep 01:29 Headers
drwxr-xr-x@  6 fritz  staff      192 11 Sep 01:29 Resources
-rw-r--r--@  1 fritz  staff  2948240 11 Sep 01:29 SDL2
drwxr-xr-x@  3 fritz  staff       96 11 Sep 01:29 _CodeSignature
Fredo:A fritz$ 
```

krl91, 2022-10-06:
sure:

```
% ls /Applications/QGroundControl.app/Contents/MacOS/../Frameworks/SDL2.framework/Versions/A/SDL2
ls: /Applications/QGroundControl.app/Contents/MacOS/../Frameworks/SDL2.framework/Versions/A/SDL2: No such file or directory
```

krl91, 2022-10-06:
I found: cast is wrong: it is SDL2.Framework and not SDL2.framework

fredowski, 2022-10-06:
I included a fix for the bug that you discovered in PR #10396. Can you try the MacOS build from there? That is: https://github.com/mavlink/qgroundcontrol/actions/runs/3197347024

fredowski, 2022-10-06:
@krl91: Can you keep this open until the fix is merged to master?

krl91, 2022-10-06:
Sure, sorry

PR #10396 (2022-10-18): MacOS: update GStreamer 1.18.1 to 1.18.6 and fix CI (#10393 / #10408) 
The CI builds on master for MacOS do not work. This PR updates the CI for MacOS and 

  * MacOS CI: Use official gstreamer package by picking patches from @patrickelectric from stable
  * MacOS CI: Update gstreamer from 1.18.1 to 1.18.6 for MacOS

With GStreamer 1.18.6 the application does not crash anymore from the videofeed

`gst-launch-1.0 videotestsrc ! x264enc ! rtph264pay ! udpsink host=127.0.0.1 port=5600`

Closes: #10393 
Closes: #10408
Closes: #10435
commit 30b594c4 (2022-10-18): MacOS: install SDL2 in directory SDL2.framework (#10435)

The build process installed the SDL2 framework at

  Frameworks/SDL2.Framework

but in MacOS 12.6 this seems to matter because the correct name
that is stored in the QGroundControl binary is

  Frameworks/SDL2.framework

This patch stores the SDL2 framework at the correct place that
fits to the location stored in the binary.

Thanks to krl91 who figured out the problem.

Closes: #10435
closed 2022-10-18 by patrickelectric