mavlink/qgroundcontrol #10437: Please handle missing/unavailable terrain data better.

AndKe, 2022-09-28:
this is my experience by current QGC and ArduCopter:

I planned a mission, and selected "Calculated above terrain"
Clicked upload ... nothing happed - no progress bar, nothing.
the upload button remains clickable, not improving the situation.
   After a long while (depending on the number of mission items) the operation times out.
I restarted QGC, same issue.
Looing at the terminal / log - I could see that it's failing to fetch terrain elevation.

It seems that the reason was unavailable terrain elevation data, and/or bad internet connection.
- Please notify the user somehow that terrain data is failing to be fetched.

AndKe, 2022-09-28:
hmm, unable to reproduce on current master - I can guess the latest appimage is outdated - or the fix was added since last appimage release.

booo, 2022-10-19:
Closing this as the author is not able to reproduce the problem anymore.

closed 2022-10-19 by booo