mavlink/qgroundcontrol #10443: Obstacle Distance Overlay showing limited FOV despite publishing to obstacle distance map bins

dirksavage88, 2022-09-29:
I am seeing this with QGC daily (linux) 506af075f, I also am publishing directly to obstacle_distance map, I am not sure why it is only showing distance readouts for that offset angle (in the screenshot) and not the other obstacle sectors (maybe a sensor fov issue?) The distance readouts in that one sector also vary enough for me to believe it's not showing the distance readouts for just that sector in the obstacle distance maps (but other sectors as well, as I have 320 degree coverage). For those familiar with obstacle distance message, you can put UINT16_MAX into the bins where coverage is either not valid or is stale data in my case (due to the sensor rotating). I have increment set to 5, # of bins 64, and angle offset set to -2.5. The rotating lidar spins from 0 to +/-160, leaving 40 degrees uncovered at the rear of the sensor.

Additionally when I run 'listener obstacle_distance_map' I am seeing the distance values in different bins as my sensor is a rotating lidar. 
[image]

Also attached is some printed error info on the terminal for qgc:
 
[image]

_Originally posted by @dirksavage88 in https://github.com/mavlink/qgroundcontrol/issues/10189#issuecomment-1262496202_

dirksavage88, 2022-12-29:
So it looks like this issue is related to the obstacle distance map when no timestamp is given to each distance array element. Added obstacle_distance timestamps and I got the expected result: 
[image]

dakejahl, 2023-11-10:
This is still an issue. I'm looking into it but I can't figure out where the +90 degrees is getting added

dirksavage88, 2023-11-10:
@dakejahl I never figured out the 90 degree offset issue. From my testing it was only an issue with the overlay and not the actual obstacle distance messages.

dirksavage88, 2025-01-08:
I believe this was fixed now in the newest stable of QGC. Closing.

closed 2025-01-08 by dirksavage88