mavlink/qgroundcontrol #10449: Can't read half of the lines in "Log Download"

dlech, 2022-10-06:
Black text on dark gray background is almost impossible to read in the Log Downloads page.

[image]

As a workaround, you can select a dark line to read it.

It would be better if the text was white.

## System Information
When posting bug reports, include the following information
- Operating System: macOS 12.6
- QGC Version: v4.2.0
- QGC build: stable
- Flight Controller: N/A
- Autopilot (with version): N/A

booo, 2022-11-06:
I can't reproduce this on Linux with latest master. Here everything is white. No alternation of the color.

Can you check a build from the master branch?

Maybe we use implicitly use some strange defaults from Qt on macOS?

dlech, 2022-11-09:
> Maybe we use implicitly use some strange defaults from Qt on macOS?

According to the docs, the default should have alternating (background) colors: https://doc.qt.io/qt-5/qml-qtquick-controls-tableview.html#alternatingRowColors-prop

I've made a PR to fix the issue.

booo, 2022-11-09:
Nice. Thanks for your contribution!

PR #10481 (2022-11-09): LogDownloadPage.qml: fix TableView text color
This passes the styleData.textColor property from the TableView to the Text items used to populate the rows to ensure the proper color of the text to match the alternating background colors of each row.

This fixes not being able to read the text on macOS where half of the rows were black text on almost black background.

Fixes: #10449
commit 5c3099e2 (2022-11-09): LogDownloadPage.qml: fix TableView text color

This passes the styleData.textColor property from the TableView to the
Text items used to populate the rows to ensure the proper color of the
text to match the alternating background colors of each row.

This fixes not being able to read the text on macOS where half of the
rows were black text on almost black background.

Fixes: #10449
closed 2022-11-09 by patrickelectric