mavlink/qgroundcontrol #10450: Unable to save logs over MAVLink

j-chen-opteran, 2022-10-07:
Hi,

I currently have an ESP8266 wired up to a Kakute H7 Mini running PX4. The Kakute H7 Mini is a v1.1, without onboard flash storage so I was counting on using ulog streaming in QGroundControl for initial testing. AFAIK the H7 mini isn't officially supported, but I've flashed it with the full size Kakute H7 target and it works. 

I'm not able to start MAVLink logging on QGC, however I suspect this might be something to do with the PX4 logger as well. 

## Expected Behavior
In QGC, MAVLink > MAVLink 2.0 logging > Start Logging should generate a log file

## Current Behavior
- The following error message occurs in QGC: `[E] at ../../qgroundcontrol/qgroundcontrol/src/Vehicle/MAVLinkLogManager.cc:935 - "Could not create MAVLink log file: """`
- In the PX4 MAVLink console, `logger status` produces the following:
```
NuttShell (NSH) NuttX-10.2.0
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 97 (3104 bytes)
INFO  [logger] Not logging
```

## System Information
When posting bug reports, include the following information
- Ubuntu 20
- QGC Version: 4.2.0
- QGC build: stable
- Flight Controller: Holybro Kakute H7 Mini v1.1
- Autopilot: PX4 v1.1.3b

gorghino, 2022-12-29:
Did you solve?

hoopertsau, 2023-08-20:
it is not included in the latest release version v4.2.8

closed 2024-05-31 by HTRamsey