mavlink/qgroundcontrol #10453: In the Firmware part the download address of ardupilot for pixhawk 4 is wrong.

FishDoggy, 2022-10-16:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

## Detailed Description
the right download address is supposed to be https://firmware.ardupilot.org/Copter/stable/Pixhawk4/arducopter.apj
however the app go to the https://firmware.ardupilot.org/Copter/stable/px4/arducopter.apj
which result to the unsuccess download.

booo, 2022-11-10:
What version of QGC do you use. Please give us some context to reproduce the problem.

FishDoggy, 2022-11-11:
Stable Release v3.5.6

booo, 2022-11-11:
Could you please try newer versions of QGC from here: https://github.com/mavlink/qgroundcontrol/releases/tag/v4.2.4

closed 2024-05-31 by HTRamsey