mavlink/qgroundcontrol #10461: ComplexItem modification

TorekV, 2022-10-24:
Hi there, 

I want to modify complex items, for example I want to add one more Loiter before my landing sequence but I do not want to add that Loiter command manually every time so I want that Loiter Item to be a part of Complex Landing Item. I reviewed the code and modified some parts but I could not manage to add any items to any complex item. 

I found out the part where complex items are written as JSON for missions but I could not manage to add my personalized item to any complex item. Can someone help me about this.

booo, 2022-10-24:
Did you try to modify the LandingComplexItem: https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/LandingComplexItem.cc#L322

TorekV, 2022-10-24:
Yes I tried it but could not get any results. Do you have an example if you added an Item to Complex Items or have a history with Complex Items

booo, 2022-10-27:
Sorry, I don't have an example for you or a history of editing complex items.

Maybe share some code that you tried so others can check what is wrong.

I'm closing this now because it is not a feature request or a bug.

closed 2022-10-27 by booo