mavlink/qgroundcontrol #10463: the compiler can't find the dylib in OSX while it is linking after I added the gstreamer into the project

kevinuav, 2022-10-25:
## Current Behavior
I use OSX(11.6) Qt(5.15.2) and Xcode(13.2.1) to compile my QGC-daily.  It can't find the dylib while it is linking after that I added the gstreamer into the path just as the instruction said.
this is the error:
```
linking QGroundControl.app/Contents/MacOS/QGroundControl
ld: file not found: @rpath/lib/libgsttag-1.0.0.dylib for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [QGroundControl.app/Contents/MacOS/QGroundControl] Error 1
```

kevinuav, 2022-10-26:
The problem solved. I downloaded the 2PKGs again via VPN(the IP address is not in China mainland). Then it works. Seems like that the gstream's PKG in china is different from the rest of the world since Chinese government controls everything comes from the liberty world.

palahmd, 2023-06-21:
Hi, could you solve this problem? I'm having the same problem.

atesahmet0, 2024-02-04:
I'm currently having the same problem.

closed 2024-05-31 by HTRamsey