mavlink/qgroundcontrol #10465: QGC does not support all sensor orientations

Jaeyoung-Lim, 2022-10-26:
## Expected Behavior
Board orientation can be configured for all values in https://mavlink.io/en/messages/common.html#MAV_SENSOR_ORIENTATION are available.

## Current Behavior
The entries in https://mavlink.io/en/messages/common.html#MAV_SENSOR_ORIENTATION  have 35 entries, but the board orientations supported by QGC only has 30 entires.
[image]

This leaves out the orientation `[MAV_SENSOR_ROTATION_ROLL_90_YAW_270](https://mavlink.io/en/messages/common.html#MAV_SENSOR_ROTATION_ROLL_90_YAW_270)`, which I needed for setting up a tailsitter

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect a autopilot to QGC and start a calibration routine

## System Information
- Operating System: [Ubuntu 18.04]
- QGC Version: [daily]
- QGC build: [daily]
- Flight Controller: [Pixhawk4]
- Autopilot (with version): PX4

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

Jaeyoung-Lim, 2022-10-27:
I noticed that for PX4, the enums are even misaligned, and therefore incorrect

booo, 2022-10-27:
Thanks for reporting the bug. The enumeration of rotations can be found here:

https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/SensorsSetup.qml#L59

Maybe you can update the enumeration and provide us with a patch?

Jaeyoung-Lim, 2022-10-27:
@booo I think the problem is more that the enumeration is done separately in QGC, rather than reading it out from mavlink definitions. However, I am not sure how to implement this properly with qmls

junwoo091400, 2022-10-27:
I can check on this later, as I am nowadays learning a lot about qt 🤔

It would be a matter of
1. C++: Fetching the orientation enums from the mavlink
2. C++: handling the name strings (not sure if we can get them directly)
3. Qml: having a checkbook be able to read the Qlist property or sth similar from the C++ side, and display it

booo, 2022-10-31:
What about adding a quick fix based on the list I referenced above? We could create a nicer fix based on the enumeration later on.

Jaeyoung-Lim, 2022-11-01:
@booo Unfortunately I wont have time to make the fix myself. 

However, it would be nice if the problem can be fixed, since using the UI results in configuring the autopilot in arbitrary attitudes that are different compared to what the UI shows.

booo, 2022-11-01:
I added the quick fix here: 

https://github.com/mavlink/qgroundcontrol/pull/10471

Can you try the build and let us know if it fixes your problem?

HTRamsey, 2024-05-31:
Seems to hav

closed 2024-05-31 by HTRamsey