mavlink/qgroundcontrol #10466: It's impossible to build qgroundcontrol.pro with Qt 6.3.1 on linux

AissataGaledou, 2022-10-28:
Hello,
Anyone have an idea on the fact that the qgc does not compile? when I build it it displays   'airmap/client.h' file not found and [Makefile:255417: QGCToolbox.o] Error 1
I'm new to this field, I tried to find a solution on the net but no change.

holden-zenith, 2022-10-30:
wrong qt version, you need 5.15.x

AissataGaledou, 2022-10-30:
Okay, i will try.

booo, 2022-11-06:
Could you provide us with some steps to reproduce your problem? Otherwise we have to close this because there is not enough information to reproduce your problem. Thanks a lot.

Davidsastresas, 2022-12-08:
Closing the issue due to inactivity and the fact that we don't support Qt 6.3.1 at the moment.

closed 2022-12-08 by Davidsastresas

zdanek, 2022-12-08:
@booo good point. According to the wiki page we're out of support. I mean we never were but time for any support means any interest from QT regarding to 5.15 version. We're at OSS plan so no interest for fixing any issues with our version of QT.

We should think of moving to QT6. I think this should be done in separate branch. 
There are two strategies. Maybe three :) 
1) branch from master and port all the code, with rebasing all MRs from master during this task - this will synchronize all the code parallelly  but it will be hard as new pieces of code will come from master. Merge back into master.
2) branch from master, port all the code. When ready, rebase from master again and port the difference. Finally merge back into master. Easier as no new code will come during porting.
3)  freeze the code on master, branch, port and merge back. This is the worst one as it stops development and bug fixing. But it's possible, so I mention it :)

Currently I'm involved into porting Google's Shaka Packager from Chromium build tools into CMake. We also get rid of "Base" which is huge set of universal toolkits in favor of C++ Abseil. Amount of work is quite big as we need to fix lot's of issues and switch to Abseil or rewrite own toolkits. We chose 2nd strategy with separate branch. In the end we'll synchronize with master, port new code and merge back into master. 

This work needs motivation and patience but it's doable and we can gain latest QT support with new features and mobile devices support.

booo, 2022-12-08:
I really don't know how much work an upgrade of the Qt version involves. Maybe we can come up with a rough estimate of the workload somehow?

I guess we should consider the availability of Qt6 in the LTS versions of popular Linux releases and other platforms too. Some LTS releases seem to not contain Qt6.

Overview of the ubuntu release cycles: https://wiki.ubuntu.com/Releases

holden-zenith, 2022-12-08:
I mentioned in the forums that I had upgraded to Qt6 for my own personal version, but removed a few things I didn't need. The biggest things from memory are that the gstreamer plugin would have to be replaced with QtMultimedia and QtLocation isn't released yet.

booo, 2022-12-08:
@holden-zenith could you share your changes with us? Would be awesome!

Davidsastresas, 2022-12-08:
So, regarding qt location, according to this:

https://bugreports.qt.io/browse/QTBUG-96795

should be almost ready. not sure if that is merged though, and I can imagine we don't want to mess with a custom version of Qt included in QGC right?

the gstreamer plugin could be more involved I think. I am no expert in video management stuff, but I think porting the current features to qtmultimedia could be really time consuming, and I am not even sure if we would have the same flexibility we have now. Isn't there any way to hook gstreamer to qt 6?

zdanek, 2022-12-08:
| Isn't there any way to hook gstreamer to qt 6?

yeah, I have the same question. Gst pipeline is quite complicated and universal at the same time so I would still stick to gstreamer. And it's very robust and performs really well with little latency with video.

Do we now some more about this why we cannot use gst in QT6? Any thoughts?

holden-zenith, 2022-12-08:
Oh actually they just started porting the gstreamer qt plugin to Qt6 like two weeks ago.
https://gitlab.freedesktop.org/gstreamer/gstreamer/-/tree/main/subprojects/gst-plugins-good/ext/qt6

zdanek, 2022-12-09:
Thank for the info.
We need to wait for this plugin.

holden-zenith, 2022-12-10:
There's some cleanup and preparation that can be done in the mean time. I can begin work on some of this if somebody will actively review the PRs

booo, 2022-12-10:
> There's some cleanup and preparation that can be done in the mean time. I can begin work on some of this if somebody will actively review the PRs

Could you sketch out what kind of cleanups and preperations you have in mind? It maybe helps other devs to understand about how much work we are talking. Sharing your past work might help too.

But I think cleanup is always nice! I'm willing to review as much as I can.

We should talk to the other devs if we have a rough plan what should happen.

holden-zenith, 2022-12-10:
There's quite a bit of code relating to old Qt bugs that is no longer relevant, unused QMLs, commented out code, unfinished code, etc. Considering how long this repo has been just a mashup of different devs' work, it's quite a bit messy. Some cleanup would make porting a little easier (such that we don't waste time porting irrelevant stuff) and make any future work easier. The actual porting to Qt6 isn't really that bad, and it'll be a lot easier now than when I did it since there's a Qt6 gst plugin now and a few other things.

holden-zenith, 2023-01-25:
from 1.22.0 "Qt6 support for rendering video inside a QML scene"