mavlink/qgroundcontrol #10480: Why doesn't iOS support serial port connection?

fanfans1, 2022-11-09:
What does iOS serial port connection need to support?

booo, 2022-11-09:
Could you please add a bit more context to your request or bug report? Thanks a lot.

fanfans1, 2022-11-10:
I use qgc code to run it on Android, which has the method of serial port connection, but when running it on iOS device, iOS device does not choose the way of serial port connection

booo, 2022-11-14:
I think QSerialPort is [missing iOS support](https://wiki.qt.io/Qt_Serial_Port#QSerialPortl) and therefore serial communication does not work in QGC.

We disable serial stuff on iOS builds here:

https://github.com/mavlink/qgroundcontrol/blob/master/QGCCommon.pri#L120

I think this is a want-fix until someone implements QtSerialPort for iOS.

booo, 2022-11-16:
I close this because I think we don't have the capacity to fix the underlying problem.

closed 2022-11-16 by booo