mavlink/qgroundcontrol #10482: Camera & MAVFtp: issues with download of camera definition file

olliw42, 2022-11-09:
## Expected Behavior

I am using QGC daily (of today) on a Win10 PC which is connected to a ArduPilot flight controller (firmware copter 4.3.0) via USB. To the flightcontroller connected is also a STorM32 gimbal controller, which provides both a gimbal and a camera component. The camera component supports the MAVLink camera service. The camera definition file is stored in the STorM32 controller and can be downloaded via MAVFtp - at least with MissionPlanner this works perfectly fine. 

With QGC, the expected behavior is that the camera defintion file (uri "mftp://storm32_camdef_dummy1_v242.xml.xz") is downloaded and digested and the respective camera gui elements created.

## Current Behavior

Instead, with QGC, it does not appear to work. In fact, it appears that MAV_MSG_FILE_TRANSFER_PROTOCOL messages are never reaching the camera component. 

(I have added lots of debug printout code to the STorM32 contoller firmware to exactly see what's going on, and with MissionPlanner the printouts are as expected, but with QGC I do not see a single printout which is related to MAVFtp, not even a single received MAV_MSG_FILE_TRANSFER_PROTOCOL message is indicated)

Note that when I make the STorM32 controller send a http url (I have stored a file on my web server) instead of the mftp uri, then it works perfectly fine, i.e. the file is downloaded, digetsed, and the expected camera gui elements are created and are working.

I note that with ArduPilot also the parameters of the flight controller are uploaded via MAVFtp, and I wonder if these two MAVFtp transactions somehow interfer. Pure speculation on my side.

## System Information
- win10
- QGC Version daily, 9.Nov.2022
- Flight Controller: MatekH743
- Autopilot (with version): Ardupilot running Copter4.3.0

## Log Files and Screenshots

[image]

olliw42, 2022-11-10:
EDIT:
I have found this comment in FTPManager.h
```
 ///     @param fromURI  File to download from vehicle, fully qualified path. May be in the format "mftp://[;comp=<id>]..." where the component id is specified.
    ///                     If component id is not specified MAV_COMP_ID_AUTOPILOT1 is used.
```
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/FTPManager.h#L35-L36

This appears to explain things. 
I quickly changed the STorM32 code to include "[;comp=105]" in the mftp uri, and the StorM32 camera component now gets the MAVFtp messages.

That's unexpected behavior however. Admittedly, the mavlink docs seem to imply this behavior. But: For the component metadata general type it is explicitely stated and required that the json file is stored on the device:

"[General metadata](https://mavlink.io/en/services/component_information.html#COMP_METADATA_TYPE_GENERAL) files must be stored on the device"

https://mavlink.io/en/services/component_information.html#file-locationsurls

(while, admittedly, in other places of teh mavlink docs one also can find weaker language like "file must be present on vehicle")

So, it looks most natural that MAVFtp is done on the device form which it has gotten the uri info in the first place.

Anyway, I guess that's not a QGC issue, but I should raise an issue in the mavlink repo to clarify the standard behavior, right?

bkueng, 2022-11-11:
This is probably not the intended behavior on the QGC side. The caller of the ftp transfer should be able to set a default comp id. And when talking to the camera, it can use the camera id. Then the url can override that via `[;comp=x]`.

olliw42, 2022-11-12:
@bkueng 
many thx for the response. good to hear that this isn't necessarily the intended behavior for the default comp id. My definition for the default compid would be that it is the same as the message's compid which provided the mavftp uri.

However, the simple work around, which I now use, is to just always specify the mftp uri with a compid :)

I btw found that the mavlink docs are quite a mess with regards to mavftp, regarding what one would imply concerning a default comp id (mavftp, camera, camp metadata), but also about e.g. the mavftp protocol itself (it seems that neither QGC nor MP does what's said in the docs).

I must admit that I'm also somewhat disappointed at that in QGC xz is not supported for camera def files, and that comp metadata seems to be only for the autopilot. Not for gimbals or cameras.

So, I guess, instead of all these can of worms I rather should be happy with the compid work-around and forget about comp metadata LOL.

I guess one should close this issue, as topic is settled, and I thus do. Pl do not hesitate to reopen if you feel like that.

:)

closed 2022-11-12 by olliw42

olliw42, 2022-11-15:
well, I made a number of attempts in this direction few years back, yielding much improved component support, which were also supported by mavlink folks, but the PRs failed nevertheless ... so, frankly, it's not so obvious what "certainly welcome" means.
anyway, made a PR adding .xz support to cam def file mftp download, #10487. Let's see.

btw, I experience some strange issues with the installation on win, which I did on twp PCs exactly as in the docs, but that's OT, jfyi.