mavlink/qgroundcontrol #10504: Mavlink inspector sees incoming packets, but QGC says it's still disconnected

DiscoMan18, 2022-12-01:
Mavlink telemetry stream (generated by iNav and sent down to ground by QCZEK LRS) is arriving to the app, and I can see every kind of packet in Mavlink inspector, but QGC says that there is still no connection. Tried every kind of links: TCP, UDP, Serial via Bluetooth. 

Tried different version of QGC on different platforms, android, iOS, Mac etc. Same thing. At the same time other app (like APM Planner 2, iNav Telemetry Viewer) work flawlessly. TCP, UDP and Serial links do work in those apps. 

What's wrong with QGC?

booo, 2022-12-04:
Please provide us with some log files and try the newest version of QGC please.

DiscoMan18, 2022-12-08:
I found out what was the issue. QGC requires sending device to identify itself as an autopilot, i.e sending HEARTBEAT message with ComponentId = 1, otherwise it won't connect. As soon as I modified the code and started to put out HEARTBEAT messages with ComponentId = 1, QGC connected and showed all the telemetry data. 

Is it a normal behaviour or a bug?

Davidsastresas, 2022-12-08:
It is expected, as otherwise it could be another mavlink device, like a datalink, or a camera.

For instance, some radio systems like rfd900 will send a heartbeat even if not connected to the autopilot, and it would be wrong if QGC detected them as a vehicle, so it is not a bug, it is how it is supposed to work.

DiscoMan18, 2022-12-08:
> Should we close the issue?

Yes please. And thank you for the feedback!

Davidsastresas, 2022-12-08:
you welcome!

closed 2022-12-08 by Davidsastresas