mavlink/qgroundcontrol #10507: New (major) release

booo, 2022-12-06:
I suggest that we release a new version of QGC that contains all recent changes to the master/development branch of the project.

I guess that should be either the v4.3.0 or v5.0.0 release.

The [newest v4.2.4 release ](https://github.com/mavlink/qgroundcontrol/releases) from Oct 21 does not contain all recent changes and e.g. we are looking for a "stable" version that supports some of the recent feature additions.

@patrickelectric maybe we can coordinate the release together? I can help wherever I can.

Davidsastresas, 2022-12-16:
I agree. It would be very cool though to have ready all those gstreamer commits @fredowski worked over recently, I think they solve important issues. I think it would be worth the effort of having them in the next release.

Maybe we should prepare a project with the stuff needed in that release, and work from there?

booo, 2022-12-28:
@mrpollo @patrickelectric @DonLakeFlyer do you have time to coordinate a stable release? I think you are most qualified because you pushed releases in the past. What is your opinion?

DonLakeFlyer, 2022-12-29:
Sorry Phillip I'm not doing that any more.

Davidsastresas, 2022-12-30:
Maybe we could prepare a new project within the repo and list in there all the matters to be checked before this release? Could be useful as well maybe to make a discord channel for this matter? Let me know your thoughts!

booo, 2022-12-30:
I don't want to join yet another chat if possible. I would prefer coordination here.

If you have some concrete bugs/changes in mind please list them here for now.

Davidsastresas, 2022-12-31:
No problem! From my part, I am working on:

- Remote ID support for QGC. It has been ready for some weeks already, we are testing it on real vehicles with real devices, etc, it should be ready soon, within a couple weeks maximum I think. I think this feature would be nice because the FAA is enforcing this very soon, so it would be great to have it on the next release specially for the US users.
- I am also working on https://github.com/mavlink/qgroundcontrol/issues/10512, which should be ready as well within 2 3 weeks I expect. I am not so interested in this one making it to stable this soon, I don't see it as important.

Besides the above, I am only aware of the work that is being developed around the video backend. I think for the next stable release we should have something clear about all of it. Let me know your thoughts please. I am open to help checking whatever we consider should go into next stable. Thanks!

booo, 2023-01-15:
We should make sure the CI pipeline supports releases. E.g. for macos needs an update for releases.

mrpollo, 2023-03-13:
We should make a list of all the PR's waiting to be merged for the release. I created a project board for tracking. Please help me add anything that should be there https://github.com/orgs/mavlink/projects/1/views/1

I went through the PR queue this morning and merged what I could, and there are still a few PRs I haven't had a chance to look into

Davidsastresas, 2023-03-13:
Thanks @mrpollo. 

I saw the video related PRs were merged recently. That together with the remote ID support ( about to be merged ), is the only features that I wanted to put in next stable from my part. I am working on some gimbal control improvements but that could go to daily if we make a stable before.

So, what everybody else thinks? Do we have everything to make a minor release 4.2.5?

booo, 2023-03-13:
Is the release pipeline ready? Do we know how to update the stores etc? Do we want to update the play stores etc.?

I suggest we tag a pre-release and test the pipeline and ask some of the users to test the pre-release.

If the release gets created from master it should be a at least 4.3 or 5.0 if we want to follow proper semantic versioning.

patrickelectric, 2023-03-14:
I agree to release a [4.3.0-rc.1](https://semver.org/) release.
There are no features for me that should hold the new release, just bug fixes and improvements, that can be released as minor versions.

mrpollo, 2023-03-14:
Let's wait for #10600 to come in and tag a `4.3-rc.1` pre-release, meanwhile I will try to come up with release notes.

DonLakeFlyer, 2023-03-14:
"I suggest we tag a pre-release and test the pipeline and ask some of the users to test the pre-release."

Follow the dev guide instructions. To create a pre-release you do exactly what you would do to create a release, you just don't update the version.txt file so everyone isn't notified about it. That should upload everything to S3 latest as wel (which is where the docs point to for stable). Then if you find problems in the pre-release you just make a point release, tag again. Once you are happy then you update version.txt and everyone then knows about the new version.

> We should make sure the CI pipeline supports releases. E.g. for macos needs an update for releases.

Why? Is there something broken?

patrickelectric, 2023-03-14:
> Let's wait for #10600 to come in and tag a `4.3-rc.1` pre-release, meanwhile I will try to come up with release notes.

I just created a https://github.com/mavlink/qgroundcontrol/pull/10622 to have a defined code standard for the project.

DonLakeFlyer, 2023-03-14:
`4.3-rc.1` I don't think the build system can deal with tags like that. Only `#.#.#` allowed. It won't create correct version numbering.

patrickelectric, 2023-03-14:
> `4.3-rc.1` I don't think the build system can deal with tags like that. Only `#.#.#` allowed. It won't create correct version numbering.

Where is such tag format limitation ?

DonLakeFlyer, 2023-03-14:
Search for the place in one of the .pri files (don't remember which one) which does a git version call. If you change this be very careful about android version numbers which don't use semantic versioning. I seem to remember they are built from the #.#.# numbers directly through some sort of math.

DonLakeFlyer, 2023-03-14:
All that said, if it ain't broke...

Davidsastresas, 2023-03-14:
@mrpollo hopefully #10600 will be fixed this week. I think I will be able to spare time on the next couple days to close it.

So, If I understood correctly, the approach is is to prepare this release from current master, and it will be 4.3 then, correct? We put together everything, but not updating version.txt, and let users start testing and report back for some days/weeks?, and if no issues are found, we update version.txt when ready so everything gets uploaded to s3. 

Is the above correct?

- Regarding macos, I checked recently and it seems to be building artifacts just fine, what is the issue?
- Regarding IOS, it's been a long time since they are not being updated, and I can imagine it will be quite a task to restore appstore uploads. I think very few users use QGC on IOS, so I would postpone this fix for the future. What do you guys think?
- Regarding Android, I remember we talked about this recently. Would it be worth it to just upload 64 bits .apk to playstore for simplicity, but update to S3 also 32 bits so the users can download it from the official QGC website? I fear the .apk bundle could eat a lot of time, and probably most users will use 64 bits anyway. What do you guys think? I checked this part and it seems so far they have been uploaded by @dagar . If Daniel helps us would be great, but if he is not available anymore for this purpose I am willing to set up the playstore myself. But, I would need Daniel confirmation about this, as we can probably not upload the same app name to play store.

Thanks!

DonLakeFlyer, 2023-03-14:
If you follow the instructions for creating a new release but don't update qgcversion.txt then this happens:
* Releases will be automatically uploaded to S3 and the links the docs refer to for Stable will be pointing to these new builds. But the next time you run previous QGC Stable you WILL NOT be prompted that there is a new version to install.
* You can ask people to download new Stables and try it out.
* Daniel got Android working initially on the store many, many years ago. Since that first time he hasn't been involved and I have been maintaining it. Getting store upload to work is a nightmare and the requirements for what it wants seems to change every couple months causing the workflows to be updated and many gray hairs to be added trying to make it work again after the Google folks broke it with new stuff. I have literally spent weeks of time keep it working until the very last time it broken and I decided I didn't want to ever touch that stuff again.
* Theoretically you could simplify things by only uploading a 64 bit apk to the store instead of a 32/64 AAB. But be prepared to wave around gallons chicken blood while you make random changes which don't seem to affect anything yet get you closer to making it work. And then a weeks worth of work later, it's still now working correctly. Do I sound bitter, yes! The amount of time I've wasted on Google Play store is monumental.
* As android currently stands I believe both 32 and 64 bits daily apks are being uploaded to S3. Unless something broke Stable apk version upload to S3 might still be working as well.
* iOs store was never automated. Gus Grubba used to do it by hand and push it up manually for each Stable. Be he is long gone.

Davidsastresas, 2023-03-15:
According to that, it might be worth then to consider fixing playstore now, or directly remove it from playstore so the users can download it from the official docs website, just like the rest of the platform. 

- So, what do you guys think, shall we drop playstore for the moment and take a look at it in the future?

Regarding IOS, we need to also make a decision about this. If we won't be maintaining IOS version for the short future, I think we should remove the older version from there. 

- Besides apple store, is there any other way for IOS users to install an app? 

Regarding the process I think it is clear for me. From the previous comments it seems the only requirement we want to include in next release is the #10600 PR. 

- So, once that is fixed, we can make the pre-release based on master. Is that fine for everybody?

DonLakeFlyer, 2023-03-15:
Others: 
* I would like to see #10603 get into stable
* #10269 seems like a major problem
* And I see a lot of noise around Windows versions not running on various machines/VMs

Davidsastresas, 2023-03-15:
@DonLakeFlyer I just added those to the list @mrpollo created:

https://github.com/orgs/mavlink/projects/1/views/1

- Should we open an issue with the Windows one? Is there any way to reproduce it? I've been building QGC lately for windows, and it is being used on production on several times I have close contact with, and I haven't heard of such issues. Please provide more details!

DonLakeFlyer, 2023-03-15:
Windows: Something here: https://github.com/mavlink/qgroundcontrol/pull/10563. And also someone in Discord was complaining. I would say do the RC nds make sure you get a decent amount of folks giving it a try on Windows. If no complaints then they are one-offs, still problems, but not enough to hold for.

Davidsastresas, 2023-03-15:
Noted. I updated the issue in:

https://github.com/orgs/mavlink/projects/1/views/1

reflecting your last comment. Is everybody else happy with the above list? What about IOS/Android, are we okey dropping IOS for the moment, and just updating android to the official QGC website?

DonLakeFlyer, 2023-03-15:
I would drop iOs from the store. You can theoretically manually push a new Stable to the Play store. I've done that before as well. It's not that hard. I would say trying to keep the store alive for Stable is worth a try. There is also a store entry for Daily. I would remove that one unless it get automated.

Davidsastresas, 2023-03-16:
@DonLakeFlyer great to know they can be uploaded to appstore manually. I guess the same applies for playstore. So far, it seems the CI itself is not an issue, but uploading it to the stores. So, considering we would be doing this not very often, maybe it is totally okey to do that manually when we have a stable release, and the rest of the time just use our website. 

Great then! Lets work over the list @mrpollo did then ( https://github.com/orgs/mavlink/projects/1/views/1 ) and lets do that pre release. Happy to see we are making progress here! Thank you everybody!

closed 2024-05-31 by HTRamsey