mavlink/qgroundcontrol #10510: Custom build ardupilot parameters tab missing

highfreq, 2022-12-09:
## Expected Behavior
Custom Ardupilot build to show parameters tab in vehicle setup

## Current Behavior
Everything compiles fine but parameters tab is not present in vehicle setup.

I compiled a custom build from today daily, i made the following changes in custom.pri to have apm plugin to be included:

# Build a single flight stack by disabling APM support
MAVLINK_CONF = ardupilotmega
#CONFIG  += QGC_DISABLE_APM_MAVLINK
#CONFIG  += QGC_DISABLE_APM_PLUGIN QGC_DISABLE_APM_PLUGIN_FACTORY

# We implement our own PX4 plugin factory
CONFIG  += QGC_DISABLE_PX4_PLUGIN_FACTORY
#CONFIG  += QGC_DISABLE_PX4_PLUGIN

I can connect to my copter but parameters tab in vehicle setup is not present.

Maybe there is a mistake in the changes i made in custom.pri or maybe it needs to be enabled somewhere and i missed it.

thanks

highfreq, 2022-12-09:
I guess i have to click 5 times the fly view button, don't know how i missed it when i read through the dev manual online.

booo, 2022-12-10:
If your problem is resolved please close the ticket.

highfreq, 2022-12-11:
Actually i am not able to have the advanced menu to pop up, i clicked 5 times pretty much everywhere.

Davidsastresas, 2022-12-16:
I think the custom build is meant to be a "guideline" of how to approach QGC modification and not having a lot of trouble when rebasing to newer QGC. I don't think it is meant to be an alternative QGC.

The base is in there, all the override system for the custom classes is in there, etc. If you follow it you will be able to grasp how to develop it exactly to fit your needs.

The custom build system allows to do a really minimal QGC version, you can even have your custom firmware plugins. It is a great tool but needs a developer to really be familiarized with the codebase. If you keep studying it you will sort it out I think.

I think we can close this one. @highfreq if you want to improve the documentation, or code for this custom QGC feature, feel free to do a PR about it!

closed 2022-12-16 by Davidsastresas

highfreq, 2022-12-16:
Thank you very much, i keep studying it. I solved the problem of the advanced not popping :)

Davidsastresas, 2022-12-16:
Yes, I am sure you will manage. From my point of view that custom build feature is not really to be able to modify QGC with no knowledge at all, because you really need to understand how it works, from my point of view the strong point of that QGC custom build system is how easier that makes future rebases. 

You can do most of the customizations over that custom build system, so your rebases in the future will be seamless, or almost seamless. Good luck!

highfreq, 2022-12-16:
Hope so, my most difficult task at the moment is transform the custom plugin from px4 to ardupilot, at the moment i included APM when compiling so it should at least be usable for me.