mavlink/qgroundcontrol #10511: Accessing pwm value of a channel

highfreq, 2022-12-09:
I need to read a pwm value of a certain channel, how do i acces it?

If for example i want to know flight mode it is in:

QGroundControl.multiVehicleManager.activeVehicle.flightMode

Is there something similar to access for example channel 12 current pwm?

Maybe my question is stupid and is it harder than that, if only someone could put me on the right path thn i can study :)

thanks in advance.

booo, 2022-12-10:
I guess you want to listen for rcChannelsChanged "events":

https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.h#L941

Maybe take a look at the Radio calibration GUI to get a better understanding on how to use the events.

closed 2022-12-10 by booo

highfreq, 2022-12-11:
Is RadioComponent.qml the radio calibration gui you are referring to?

booo, 2022-12-11:
`git grep Vehicle::rcChannelsChanged` is your friend.

highfreq, 2022-12-11:
I took a look at the method, please a little more help, i am an old guy fighting to learn :)

If i want for example to show current pwm value of channel 3 on a label what would i have to write on label text

QGroundControl.multiVehicleManager.activeVehicle.rcChannelsChanged()

thanks in advance.

highfreq, 2022-12-11:
Ahahahah, thanks.

highfreq, 2022-12-13:
Why did you close it, maybe someone else will have time to help me out

booo, 2022-12-13:
Because you got a pointer in the right direction from me and the developer team does not have time to answer questions and help in the development of proprietary code. If you have plans to contribute to QGC please publish your code and open a PR. We are more than happy to review contributions to QGC.

highfreq, 2022-12-13:
Not only devs hang around here, maybe someone else would peek in and help me out. Anyway thank you.