mavlink/qgroundcontrol #10513: Enter advanced mode on custom build

highfreq, 2022-12-10:
I tried to enter advanced mode in my custom build but looks like nothing works.
Tried 5 taps on version, 5 taps on blank toolbar. It doesn't work.

Any hint.

Thanks.

zdanek, 2022-12-10:
Enable all logs and send them here please.

highfreq, 2022-12-10:
How do i enable all logs?

What would be the normal procedure to enable advanced mode?

highfreq, 2022-12-11:
I think there is something wrong on the onClicked easter egg in MainRootWindow.qml. If i remove all the if statements and just open advancedmodeconfirmation than it works. For the advanced mode i guess it is this condition that is not met mouse.modifiers & Qt.ShiftModifier

mentonin, 2023-10-23:
The condition is literally a shift + click. Which I believe can only be achieved with a keyboard connected. Probably should be changed to a double click or something. And documented on the wiki.