mavlink/qgroundcontrol #10518: Repo Cleanup

holden-zenith, 2022-12-11:
General Discussion

holden-zenith, 2022-12-11:
@booo what is the state of iOS integration? Is it still functional?

booo, 2022-12-15:
> @booo what is the state of iOS integration? Is it still functional?

I have no expertise in this area.

> General Discussion

I guess this needs a bit more of explanation :)

Davidsastresas, 2022-12-16:
What do we mean here for repo clean up? that is very broad!

Regarding IOS, what is the current status? Is there something not building or not working for IOS or we just don't have the appstore thing working?

If it is the second, If I recall correctly, it needs to be done manually, I think we can not do that in the CI. At least it was like that last time I looked into it, a couple years ago.

booo, 2022-12-16:
@Davidsastresas or @fredowski can you summarize which commits need merging for the video support?

Davidsastresas, 2022-12-16:
I started to get familiarized with it very recently, fredowsky have definitely more criteria than me to answer here.

closed 2024-02-19 by HTRamsey