mavlink/qgroundcontrol #10519: License

highfreq, 2022-12-16:
If requested the source code of a modified QGC, is the person/company who modified it forced to release it under the apache/gpl that QGC applies?
I read both of them but i don'tunderstand honestly.

zdanek, 2022-12-16:
IMO yes.
Anybody wise to answer the question?

Davidsastresas, 2022-12-16:
Not always.

The GPL license means if you purchase something with modified QGC on it, and that modified QGC was developed over GPL license ( not paying fees to Qt ), you would be in your right to request the code. But only if you purchased something with modified QGC on it. You can not approach a random company you suspect has modified QGC and just request the code.

The Apache license ( if that modified QGC was developed/distributed paying the correspondent fees to Qt ), allows to closed the source code, so if the company used this license instead you have no right at all to request the code, even if you purchased the product, it is virtually like closed source they develop, from the end user point of view.

Just out of curiosity, which is that custom QGC you are interested in? If you are interested in a particular feature maybe we could open an issue about it to get it developed in QGC.

closed 2022-12-16 by Davidsastresas

highfreq, 2022-12-16:
https://www.youtube.com/watch?v=sqVFtMFgGbI&feature=emb_rel_end

I don't think, they paied anything to anyone, maybe you could ask. They implemented touch controls, similar to what we were talking in the other thread.

Davidsastresas, 2022-12-16:
In that case you probably can ask them.

Although it seems odd to me. In that video QGC reports disconnected. I don't know, it doesn't seem like a good integration to me. As you say, what we are talking about in the other thread should cover this, I personally don't think it is worth the effort of even asking them.

Please tell me more about how it works, regarding which mavlink messages they use to control etc etc.

highfreq, 2022-12-16:
That is why i would like to see the code, not to implement like they did, but to get a starting point. Randy from the other thread where we talked about it knows a lot. He is the one that brought mavlink support to their gimbal.