mavlink/qgroundcontrol #10523: "Crash dump on SD, vehicle needs service"

Attahayat, 2022-12-19:
I am using Pixhawk 5 with PX4 installed on it. My operating system is Ubuntu 20.04. When I try to arm my rover, I get an error "MAV_CMD_COMPONENT_ARM_DISARM temporarily disabled". I disabled all the prearm checks that could have caused this but I am still getting it.

When I go to the safety tab I get an error that my firmware does not have the param CP_DIST which after looking up on here, I found out that it might be caused because the power is not set up correctly. I set up the batteries again (power module) and calibrated them but I still get the error. Also on the main screen, I am notified that there is some crash dump on SD, and the vehicle needs service which I do not understand the meaning of.

I tried to find what might be the cause here but I cannot find anything. Any help would be appreciated as this has been very frustrating. Thanks.

Davidsastresas, 2022-12-19:
Hello @Attahayat . It seems you have an issue with PX4 setup, not with QGC. You will have better luck asking in the dedicated PX4 forums or chats. I would be happy to help but I am not familiarized with PX4, only with Ardupilot, so I can not be of much help.

Attahayat, 2022-12-21:
Thanks, it has something to do with the updated px4 firmware with microdds_client installed on it. It worked fine previously with micrortps_client. I will close it here.

closed 2022-12-21 by Attahayat

abolfazlo, 2023-06-29:
> I am using Pixhawk 5 with PX4 installed on it. My operating system is Ubuntu 20.04. When I try to arm my rover, I get an error "MAV_CMD_COMPONENT_ARM_DISARM temporarily disabled". I disabled all the [...]

I had the same error. I removed the log files under this directory: /fs/microsd, and it worked.