mavlink/qgroundcontrol #10526: Voltage/Amper Divider

zehrasevinc, 2022-12-24:
When I try to calculate voltage divider and amper divider the calculate button that I mark on picture is doesn't work. I guess that's why I can't do ESC calibration and that's why motor's each motors working seprate.

- Operating System: Windows 11 pro
- QGC Version: 4.2.4
- QGC build: Stable
- Flight Controller: Pixhawk Cube Orange
- Autopilot (with version): PX4 1.13.2

[image]

booo, 2022-12-28:
What kind of power module do you use? Did you configure the right power module?

This should not affect your ESC calibration etc.

DonLakeFlyer, 2023-02-18:
As you can see from the screen shot you vehicle is not sending voltage to QGC (Vehicle voltage). No way to calc a divider without that. THis is a vehicle hardware problem.

closed 2023-02-18 by DonLakeFlyer