mavlink/qgroundcontrol #10527: Linking ROS2 with QGC

serkanMzlm, 2022-12-28:
I managed to connect ROS2 and QGC together ubuntu 20.04. but when I got the system to windows environment, I tried to open the qmake file via QT. QGC worked (without including ros2).  Error when trying to open it via Qt with cmake.  It was fixed when I corrected the `${CMAKE_CURRENT_SOURCE_DIR}/windows/QGroundControl.rc)`  code in the WIN32 control area as
`${CMAKE_CURRENT_SOURCE_DIR}/deploy/windows/QGroundControl.rc)` code.  After the compilation process was finished, the first part played and then it started to give an error. 
(I tried combining it with ROS2 for ubuntu 20.04 and it worked fine.)

OS: Windows 10
ROS: foxy
----------------------------------
OS: Ubuntu 20.04
ROS: galactic
--------------------------------

Note: It works when including ubuntu ros, but normal cmake file does not work on windows

[image]

zdanek, 2022-12-28:
Hi.
What is the usecase of connecting ROS with QGC?
Best,
Bartek

booo, 2022-12-28:
The time of our developers is very limited and we can't help with such questions. If you plan to make a open source/free software contribution to QGC it is much more likely that a developer finds time to help you.

closed 2022-12-28 by booo

serkanMzlm, 2022-12-28:
> Hi. What is the usecase of connecting ROS with QGC? Best, Bartek

I made the connection part with ROS2, so I will make communication between swarm and drone easier and communication between stations easier. 
I did it on Ubuntu side.
It started giving an error after switching to Windows.
I have never touched the codes on the windows side, the same codes on the github page 
When I open the .pro extension with QT, QGC works, but when I open it with cmake, it gives an error 

`${CMAKE_CURRENT_SOURCE_DIR}/windows/QGroundControl.rc)` code in the WIN32 control area as
`${CMAKE_CURRENT_SOURCE_DIR}/deploy/windows/QGroundControl.rc) `code.
this is how i solved the problem  

The build process finished successfully, but it gives the above error while running

serkanMzlm, 2022-12-28:
> The time of our developers is very limited and we can't help with such questions. If you plan to make a open source/free software contribution to QGC it is much more likely that a developer finds time to help you.

QGC does not work in windows environment when run with the cmake file of the problem. After the build process is finished, I get the error I mentioned above
Unless a fix is ​​made in the part I mentioned, QT does not detect it. Is this fix a bug?

serkanMzlm, 2022-12-28:
`git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git`   When I try to open it directly in the QT program without making any changes, it gives an error. (CMakeLists.txt)

serkanMzlm, 2024-04-03:
When I started ROS2 via cmd on Windows environment and opened QT from the same terminal, it worked.

Jimi1811, 2024-07-25:
> When I started ROS2 via cmd on Windows environment and opened QT from the same terminal, it worked.

Hello @serkanMzlm. I am a student and I am interested in use this platform with ROS2. Any recommendation for this would be really helpful!

serkanMzlm, 2024-10-04:
> Hello @serkanMzlm. I am a student and I am interested in use this platform with ROS2. Any recommendation for this would be really helpful!

Bunun için aklıma iki yöntem geldi. Birincisi ros2'deki konuları direkt mavlink girişlerine bağlayabilirsiniz, ya da cihazdan gelen verileri ros2 ile alıp mavlink'e çevirebilirsiniz.