mavlink/qgroundcontrol #10529: CI for android is not uploading to play store

Davidsastresas, 2023-01-02:
It has been like this for I while I think, both for stable and daily. In play store it seems to be uploaded by @dagar . Is there anything we can do to bring them back online?

booo, 2023-01-07:
Duplicate of https://github.com/mavlink/qgroundcontrol/issues/10207

Davidsastresas, 2023-01-08:
I think it was a misunderstanding closing this one and reopening #10207. 

If I recall correctly, we used to use travis to upload QGC to the play store, so it is possible that it got lost when we changed to github actions.

I am not sure right now of how we can include this in github actions, but I will take a look in the following weeks. I will report back.

booo, 2023-01-08:
> I think it was a misunderstanding closing this one and reopening #10207.

I don't think it is a misunderstanding. The user made a clear reference to the playstore in the ticket. But I don't care where we keep track of the issue.

The travis build contains code for the playstore upload: https://github.com/mavlink/qgroundcontrol/blob/master/.travis.yml#L225

Davidsastresas, 2023-01-09:
You are right, my apologies, I checked this real quick and I was paying attention to the PR #10208, not to the issue  #10207, you are totally right.

Coming back to the matter, Thanks for the link, I will play around and try to fix it in the following days. If we want to keep using the same account ( Daniel Agar ), I think we should contact him, I mentioned him on the first message.

In the meantime I will do the tests over github actions over my google play account. When it is fixed I will report back to see how we can proceed regarding which account we use.

closed 2026-10-05 by github-actions[bot]