mavlink/qgroundcontrol #10530: QGC does not work correctly when built with cmake on windows

ukb1, 2023-01-03:
OS: Windows 10
QT Version: 5.15.2
Hi,
I tried to compile with both qmake and cmake on QGC Qt Creator.
When I compiled it with qmake, the codes compiled correctly. But when I try to compile with cmake it gives an error.
linux operating system is running. I think it's the bottom part.
`
if (WIN32)
	# append application icon resource for Windows
	set(QGC_RESOURCES
		${QGC_RESOURCES}
                **${CMAKE_CURRENT_SOURCE_DIR}/windows/QGroundControl.rc)**
endif()
`

Since there is no such path, it gives an error in the add_executable part. When switching to the win operating system, this part is opened.

 **${CMAKE_CURRENT_SOURCE_DIR}/windows/QGroundControl.rc)** path should be fixed to **${CMAKE_CURRENT_SOURCE_DIR}/deploy/windows/QGroundControl.rc)** so that cmake will run

Although it doesn't work as well as the result of compiling with qmake, at least cmake works on compiling.

Can this situation be controlled?

booo, 2023-01-07:
Please provide us with the exact steps you used to produce the problem. Otherwise it is super hard to tell if there is a bug in your execution or in the QGC. Thanks a lot.

ukb1, 2023-01-10:
> Please provide us with the exact steps you used to produce the problem. Otherwise it is super hard to tell if there is a bug in your execution or in the QGC. Thanks a lot.

OS:                     Windows 10
QT Version:        QT 5.15.2
Cmake Version:  3.18.2

Note:  These codes work on linux. both cmake side and qmake side. The file works when you create it with qmake on Windows OS. But when I build it with cmake, it gives an error.

step 1:   download file  (    `git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git` )
step 2: I open the downloaded file with the Qt creator application. I open the downloaded file with Qt creator and it gives error (CMakeLists.txt).  Error in main cmake file. the cause of the error is that the file paths are not correct 

wrong code
`
${CMAKE_CURRENT_SOURCE_DIR}/windows/QGroundControl.rc)
`

corrected  code 

`
${CMAKE_CURRENT_SOURCE_DIR}/deploy/windows/QGroundControl.rc)
`

when i fix this part the error goes away.
[image]
 After the build process is finished, the file does not work because there is a missing dll file. After completing the dll files, QGC runs. The only problem is the icon is not visible 
cmake build = 
[image]

qmake build = 
[image]

The only part I can't figure out is why this icon doesn't appear

closed 2024-05-31 by HTRamsey