mavlink/qgroundcontrol #10535: Fixed wing servo setup

jorgeneo560, 2023-01-07:
Tell us a bit about the feature:
- What problem does it solve?
Make a way for easy setup for fixed wing aircracft with multiple servos
- What does it do?
Allow pixhawk firmware to be user frendly configurable
- What flight stacks must it work with? (All, PX4, ArduPilot)
PX$
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
Fixed wing
- Are there other systems that have this feature?
-Maybe
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)
Please make a easy way to configure servo control surfaces, i was ending using ardupilot on missionplanner, its very hard to configure a plane that is not the standar 4 channel, my aircraft has 2 alieron, 2 flaps, 1 aerobrake, 1 retractable, 1 inverted vtail, 1 servo on the trhottle engine

booo, 2023-01-07:
Please provide us with some descriptions of the problems you encountered and the ways you would like them to improve.

jorgeneo560, 2023-01-09:
this screen will be a nice addon below the radio calibration
[image]

closed 2026-10-04 by github-actions[bot]