mavlink/qgroundcontrol #10536: QGC Crashes on Mac OSX Ventura

EG-Julien, 2023-01-10:
Hi,
I'm using QGC on Mac OS since quite a while now and I have some problems with the map loading when I start QGC.
It tries to load the map from the cache and put my laptop location ont it. if it fails QGC is becoming unresponsive.

[image]

If I try to launch it through my terminal I got this :

```bash
./QGroundControl
Command line interface to a user's defaults.
Syntax:

'defaults' [-currentHost | -host <hostname>] followed by one of the following:

  read                                 shows all defaults
  read <domain>                        shows defaults for given domain
  read <domain> <key>                  shows defaults for given domain, key

  read-type <domain> <key>             shows the type for the given domain, key

  write <domain> <domain_rep>          writes domain (overwrites existing)
  write <domain> <key> <value>         writes key for domain

  rename <domain> <old_key> <new_key>  renames old_key to new_key

  delete <domain>                      deletes domain
  delete <domain> <key>                deletes key in domain

  import <domain> <path to plist>      writes the plist at path to domain
  import <domain> -                    writes a plist from stdin to domain
  export <domain> <path to plist>      saves domain as a binary plist to path
  export <domain> -                    writes domain as an xml plist to stdout
  domains                              lists all domains
  find <word>                          lists all entries containing word
  help                                 print this help

<domain> is ( <domain_name> | -app <application_name> | -globalDomain )
         or a path to a file omitting the '.plist' extension

<value> is one of:
  <value_rep>
  -string <string_value>
  -data <hex_digits>
  -int[eger] <integer_value>
  -float  <floating-point_value>
  -bool[ean] (true | false | yes | no)
  -date <date_rep>
  -array <value1> <value2> ...
  -array-add <value1> <value2> ...
  -dict <key1> <value1> <key2> <value2> ...
  -dict-add <key1> <value1> ...
Settings location "/Users/julien/.config/org.qgroundcontrol/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-FR"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/Users/julien/Documents/QGroundControl/Logs"
Map Cache in: "/Users/julien/Library/Caches/QGCMapCache300" / "qgcMapCache.db"
qt.qpa.fonts: Populating font family aliases took 72 ms. Replace uses of missing font family "Open Sans Semibold" with one that exists to avoid this cost.
qml: QGCCorePlugin(0x600003c8f250) []
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
QGCPositionManager error 0
"The operation couldn’t be completed. (kCLErrorDomain error 1.)"
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
2023-01-10 16:22:50.584 QGroundControl[65812:6719305] *** -[AVCaptureSession stopRunning] stopRunning may not be called between calls to beginConfiguration and commitConfiguration
```
----

## Expected Behavior
QGC shouldn't crash and location services permission should not change each time I launch QGC.

## Current Behavior
At each start of QGC, location permission are reset to deny and it makes QGC unresponsive.

## Steps to Reproduce:
I don't have identified steps to reproduce yet ... Sometimes it works and sometimes it doesn't want to ...

## System Information

- Operating System: Mac OS Ventura (M1 MAX)
- QGC Version: latest (4.2.4)
- QGC build: stale
- Flight Controller: PixHawk 4
- Autopilot (with version): PX4 1.13.2

## Detailed Description
QGC needs location permission to locate himself on the map. When QGC starts, it try to locate the laptop but at this exact timing, the permission is reset and set to deny. It makes QGC crash.

booo, 2023-01-11:
Did you check if the daily builds work for you? You can find them as artifacts of the github actions somehwere here:

https://help.mikrotik.com/docs/pages/viewpage.action?pageId=141197388#heading-Setup

EG-Julien, 2023-01-16:
It seems that installing daily builds reset permissions in Mac OS X but the problems still occurs at some random moments.
When it happens I need to download daily build and install it again to get it working...

I didn't find any infos about steps to reproduce yet ... 

*Is the link you gave related to QGC ? It seems to be a documentation for a router ?*

booo, 2023-01-16:
> Is the link you gave related to QGC ? It seems to be a documentation for a router ?

Haha. Sorry. This was the link I wanted to share: 

https://github.com/mavlink/qgroundcontrol/actions/workflows/macos_release.yml

Maybe you can try disabling the location services to see what happens:

https://macpaw.com/how-to/enable-disable-location-services-mac

EG-Julien, 2023-01-16:
Np for the link!

I'll try that at next crash.

EG-Julien, 2023-02-10:
Somehow, deleting cached files and .config/org.qgroundcontrol seems to reset location permissions.

```bash
rm -rf ~/.config/org.qgroundcontrol/
sudo rm -rf /Users/$USER/Library/Caches/QGCMapCache300/
sudo rm -rf /Users/$USER/Library/Caches/QGroundControl.org
```

I have tried to use tccutil to reset all permissions but it seems to be broken in Mac OS Ventura ... thanks Apple ...

booo, 2023-02-12:
Do you think we can do something in QGC to mitigate the problem? Maybe we can catch an error and report it to the user?

Or should we close this as resolved?

EG-Julien, 2023-02-13:
I don't know if we can catch the error, it seems that it can be returned if someone decide to deny localisation access as mentioned here :
> "The operation couldn’t be completed. (kCLErrorDomain error 1.)"

But, if we can catch the error properly, at least a textbox showing that something is going wrong with an error code and a hint to these commands should be more than enough imo.

In the other hand, we should try to find why Mac OS is reseting localisation access randomly.

Geko580, 2024-06-24:
After update to QGC 4.4 I have the same problem, screen frozen at launch on MacOS Ventura 13.6.7, and no possible action with mouse.
That's pity to don't use any more... I tried to launch QGC under Windows 10 via Parallels, but that's also don't run...

closed 2026-10-05 by github-actions[bot]