mavlink/qgroundcontrol #10541: QGC unable to detect serial ports on latest version.

justvijit, 2023-01-13:
Hi all, I am having a strange issue. So I have Skydroid H12, and when I try to connect it to my CubeOrange, it never connects. So I spoke to skydroid people and they gave a custom QGC apk (4.1.7) and gave me instructions to create a new comm link on serial port ttys1 and 921600 baud rate. And it started working. So I went ahead and tried the same thing in latest APK, and I could see that it always shows "No Serial Ports Available" in all the versions new or old. 

Since I have to customise QGC and add custom code to it, and I do not have any info or code of whatever customisation skydroid people did on the APK they gave me, I am at a deadlock. 

Definitely something to do with Skydroid blocking QGC from accessing the list of available ports, but since they gave me a QGC custom build, which is very easily able to connect, I am hoping this issue can be solved with some code tweaks in QGC.

Please help, I have been banging my head in the issue from the past two weeks.

## Expected Behavior
1) QGC on Android on Skydroid should connect automatically.
2) If not, then it should display a list of available Serial Ports inside Comm Links section.

## Current Behavior
1) Never connects automatically on Serial or UDP connection. One solution suggested by someone was to use a background app that enables QGC to connect to Ardupilot via UDP on 14551, but this requires running the app constantly in background, and is not a very reliable solution.
2) Never displays available serial ports, no matter what I do.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Well I have 10 skydroid H12 and this is happening on all of them. 

## System Information
When posting bug reports, include the following information
- Operating System: Android 8.1 (Cant be updated further)
- QGC Version: 4.2
- QGC build: Stable 4.2
- Flight Controller: Cube Pilot
- Autopilot (with version): Ardupilot Copter 4.3

Davidsastresas, 2023-01-13:
I think the fastest route to solve this is to get Skydroid to let us know why the standard apk is not working on their GCS. It could be really time consuming to debug otherwise, and it is on their interest to have this solved for everyone!

justvijit, 2023-01-14:
Hi @Davidsastresas , Thanks for replying. Skydroid people suck. It took around a month of followups to get the version they sent. Now I have no response from them. Also, its a covid situation in China along with Chinese New Year, so I not expecting a response for atleast a month or so.

Davidsastresas, 2023-01-14:
Sorry to hear that... Maybe somebody knows about this in Ardupilot or PX4 forums. If you find out more report back here so hopefully we can take a look at it. I am sorry but without knowing more about it or having a device I can not do much more.

booo, 2023-01-15:
As far as I can see Android 8.1 is end of life. Security updates stopped in October 2021. I don't think that is a platform we want to support as devs.

If the link to the UAV is a serial connection everything should work just fine. If proprietary code is involved we can't do much about it. Implementing custom code for proprietary hardware/software combination is again out of the scope of the QGC project and devs. Our time is super limited. Sorry.

Maybe you can get some kernel logs and logs from the android system via adb?

I'm not sure how the dual licensing exactly works but maybe you can ask them to release the changes?

booo, 2023-01-15:
I'm closing this because it's out of the scope of this project and it's human resources.

closed 2023-01-15 by booo

sinkyungbaek, 2023-09-19:
https://www.youtube.com/watch?v=4clMpeZHNCg

purnimawbe, 2025-01-28:
Hi also need help with QGC 4.1.7. Can you please send me the link to apk