mavlink/qgroundcontrol #10552: Decrease compiled binary file size in Linux

harshmittal2210, 2023-02-01:
Hi

The tool is really awesome and I am able to modify things easily in it. My only doubt is after deploying the binary size comes out to be large. So if I download it from the official website it is somewhere around 80MB, but mine is around 400MB. 

I have checked for both release and debug modes. Do I have to add any new configuration in the Qt compilation or when I am creating via terminal?

As of now, I am using the following commands:

```bash
cd deploy
./create_linux_appimage.sh ../ <release_staging_dir>
```
OS: `Linux`
Branch: `master`

It would be great if any documentation could be provided for the same as I am not able to find any.

Thanks in advance!

booo, 2023-02-05:
I tried building AppImages with linuxdeploy today because the stable AppImage is missing the ssl library and does not work on my bleeding edge ArchLinux system. At least linuxdeploy tends to include quite a lot of dependencies. The tool checks the QGC binary and starts to included all shared libraries in a recursive fashion. I guess thnumber of libraries you have installed on your development machine can have an impact on the size of the appImage if for some reason the tools think they should be included.

Do you have any output from the script?

harshmittal2210, 2023-02-08:
Hi @booo 

Is there a way to limit those dependencies?

I am attaching the latest log here:

```bash

sending incremental file list
NAN.txt
QGroundControl
bound_test.plan
bound_test_plan.plan
qgroundcontrol-start.sh
qgroundcontrol.desktop
qgroundcontrol.png
qt.conf
temp.pem
temp1.pem
test
test1.txt
Qt/
Qt/libs/
Qt/libs/libQt5Charts.so.5
Qt/libs/libQt5Charts.so.5.stamp
Qt/libs/libQt5Core.so.5
Qt/libs/libQt5Core.so.5.stamp
Qt/libs/libQt5DBus.so.5
Qt/libs/libQt5DBus.so.5.stamp
Qt/libs/libQt5Gui.so.5
Qt/libs/libQt5Gui.so.5.stamp
Qt/libs/libQt5Location.so.5
Qt/libs/libQt5Location.so.5.stamp
Qt/libs/libQt5Multimedia.so.5
Qt/libs/libQt5Multimedia.so.5.stamp
Qt/libs/libQt5MultimediaQuick.so.5
Qt/libs/libQt5MultimediaQuick.so.5.stamp
Qt/libs/libQt5Network.so.5
Qt/libs/libQt5Network.so.5.stamp
Qt/libs/libQt5OpenGL.so.5
Qt/libs/libQt5OpenGL.so.5.stamp
Qt/libs/libQt5Positioning.so.5
Qt/libs/libQt5Positioning.so.5.stamp
Qt/libs/libQt5PositioningQuick.so.5
Qt/libs/libQt5PositioningQuick.so.5.stamp
Qt/libs/libQt5PrintSupport.so.5
Qt/libs/libQt5PrintSupport.so.5.stamp
Qt/libs/libQt5Qml.so.5
Qt/libs/libQt5Qml.so.5.stamp
Qt/libs/libQt5QmlModels.so.5
Qt/libs/libQt5QmlModels.so.5.stamp
Qt/libs/libQt5QmlWorkerScript.so.5
Qt/libs/libQt5QmlWorkerScript.so.5.stamp
Qt/libs/libQt5Quick.so.5
Qt/libs/libQt5Quick.so.5.stamp
Qt/libs/libQt5QuickControls2.so.5
Qt/libs/libQt5QuickControls2.so.5.stamp
Qt/libs/libQt5QuickShapes.so.5
Qt/libs/libQt5QuickShapes.so.5.stamp
Qt/libs/libQt5QuickTemplates2.so.5
Qt/libs/libQt5QuickTemplates2.so.5.stamp
Qt/libs/libQt5QuickWidgets.so.5
Qt/libs/libQt5QuickWidgets.so.5.stamp
Qt/libs/libQt5SerialPort.so.5
Qt/libs/libQt5SerialPort.so.5.stamp
Qt/libs/libQt5Sql.so.5
Qt/libs/libQt5Sql.so.5.stamp
Qt/libs/libQt5Svg.so.5
Qt/libs/libQt5Svg.so.5.stamp
Qt/libs/libQt5Test.so.5
[... 1859 lines cut ...]
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
/tmp/tmp.CQAz3gG309/QGroundControl.AppDir should be packaged as /tmp/tmp.CQAz3gG309/QGroundControl.AppImage
appimagetool, continuous build (commit effcebc), build 2084 built on 2019-05-01 21:02:41 UTC
Using architecture x86_64
Deleting pre-existing .DirIcon
Creating .DirIcon symlink based on information from desktop file
WARNING: AppStream upstream metadata is missing, please consider creating it
         in usr/share/metainfo/qgroundcontrol.appdata.xml
         Please see https://www.freedesktop.org/software/appstream/docs/chap-Quickstart.html#sect-Quickstart-DesktopApps
         for more information or use the generator at http://output.jsbin.com/qoqukof.
Generating squashfs...
Embedding ELF...
Marking the AppImage as executable...
Embedding MD5 digest
Success

Please consider submitting your AppImage to AppImageHub, the crowd-sourced
central directory of available AppImages, by opening a pull request
at https://github.com/AppImage/appimage.github.io

```

booo, 2023-02-08:
I'm sorry, but I'm really no expert in this. I think our current AppImage is "broken" and I will try to fix this as soon as I have time and if my skills permit.

I tried to follow these instructions about AppImage creation with linuxdeploy:

https://docs.appimage.org/packaging-guide/from-source/native-binaries.html

With linuxdeploy you actually see what dependencies/shared libraries from the system get copied into the AppImage. Or if you do `ldd QGroundControl` on the QGC binary you already see that QGC can have quite a few dependencies.

If you don't need some of the QGC features e.g. video streaming support you can try to disable the feature and reduce the dependencies this way I guess.

harshmittal2210, 2023-02-08:
Sure, will try on my end as well. If I make any progress I'll update here.

Thanks for your help!

smagellan, 2023-02-08:
There is a way to reduce the size:
```
du -h -c qgroundcontrol/ | grep total
46M     total
```
You need to use system Qt libs + run the strip utility over QGC to achieve this

harshmittal2210, 2023-02-22:
Hi @smagellan 

Sorry, somehow I missed your comment.

I am a little new to this so could you explain a little more or share any tutorials that I can follow?

When I run it on my system I am getting:

```bash
du -h -c qgroundcontrol/ | grep total
1.2G	total
```
Also, I have kept my build folder inside `qgroundcontrol` folder itself, maybe that is why I have a very large size showing. 

I have installed all the Qt libs in my system and things are compiling perfectly fine. After running the `./create_linux_appimage.sh` it is copying different qml libraries. Are you referring to eliminating that?

booo, 2023-02-22:
Try to run the `strip` binary on the QGC binary before you create the AppImage.

smagellan, 2023-02-23:
@harshmittal2210, yes, build folder is huge: 
```
du -hc qgroundcontrol-build | grep total
1.7G    total
```

I do not build  AppImage. Instead, I use a regular DIY build: https://dev.qgroundcontrol.com/master/en/getting_started/CentOS.html, and then:
1. Unpack package/QGroundControl.tar.bz2 into separate directory.
2. Remove  everything except libairmap-cpp.so* from Qt/libs
3. Remove qt.conf
4. Strip QGroundControl binary

These steps are relatively safe given:
1. You have system-wide QT libs
2. You do not update system-wide QT libs after build (QGroundControl uses some private QT headers. I guess it may break QGC if you update QT libs).

harshmittal2210, 2023-02-24:
Hi @booo 

After running `strip` command I could reduce the binary size from 480MB to about 250MB. Should I also enable any flags while running it to decrease the size further? Just wondering about how official builds are less than 90MB.

harshmittal2210, 2023-02-24:
@smagellan  Sure I'll try to implement the same for the Linux build as well. Thank You!

closed 2024-05-31 by HTRamsey