mavlink/qgroundcontrol #10557: Survey produces no mission items if Mission start specifies "Terrain Frame"

AndKe, 2023-02-02:
## Expected Behavior
Survey grids should work equally _ regardless of altitude frame.

## Current Behavior
Survey grid is not generated (as output during upload) if terrain frame is selected  

## Steps to Reproduce:
please see video: https://drive.google.com/file/d/1cO4LV8ajPKtE2si55mNuFMI5NBMEyVZg/view?usp=sharing

## System Information
- Operating System: Ubuntu22.10
- QGC Version: 
- QGC build: current HEAD
- Flight Controller: SITL
- Autopilot (with version): Copter, current release.

## Detailed Description
- video says all.

AndKe, 2023-02-02:
update: This is not really related to mission start, but rather the altitude frame of the survey:
[image]

AndKe, 2023-05-04:
@Davidsastresas how is your schedule?  - back in February I think this was #6  :)

Davidsastresas, 2023-05-07:
Sorry for taking this long @AndKe. I was still busy with other ongoing projects but I spared some time to fix this one:

10678

It turns out to be a bug where QGC was expecting to have terrain data on the surveys if these had terrain frame. The behaviour you showed in your video was only present when this terrain info was not present, with terrain info the survey was generated correctly, so I pulled that string and it turns out that was the reason.

Thank you very much for reporting this, and sorry for taking this long to address it. Let me know how it goes, and if it fixes the problem for you please close the issue and comment on the PR. Thanks!

closed 2026-10-05 by github-actions[bot]