mavlink/qgroundcontrol #10559: GstVideoReceiver does not handle NV21/NV12/YUV420 RTSP Streams Corretly

mrawding, 2023-02-03:
Hi all, Seeking help/guidance to solve this issue on my own and submit a PR since I know the QGC developers are not Gstreamer developers. Please read below.

The Gstreamer pipeline is not handling nv21 streams from rtsp servers correctly.

The displayed image on the qtsink has the width/height flipped and cuts out the color.

The image should look like this:

 Running rtsph264 or rtsp265 server that encodes and outputs frames in YUV420 OR NV21 and viewing the stream with:

```ffplay rtsp://<ip_address>:8000/live```

yields the image below (verifies server is functional)

[image]

QGC shows this when displaying the stream

[image]

I've tried adding a ```videoconvert``` gstElement in between ```_decoder``` and ```_videoSink``` in ```GstVideoReceiver.cc```

at function ```GstVideoReceiver::_addVideoSink()```

It was still functional, but did not produce the desired image. I tried adding an RGB capsfilter too after videoconvert, but that shutdown the pipeline. 

I have a decent knowledge in gstreamer and am happy to find a solution and create a PR. I just ask that I may be pointed in the right direction!

 Clearly all of the information is there since it is rescaling and transforming the image with height->width and width->height.

Since it is NV21 the stream's color is at the end of the buffer and is being dropped. 

I can confirm that this is a problem on LINUX, but iOS displays correctly at the moment.

booo, 2023-02-05:
A few improvements to the GStreamer code were proposed here:

https://github.com/mavlink/qgroundcontrol/pull/10420

Maybe that helps somehow.

mrawding811, 2024-04-03:
This issue is resolved by upgrading on Linux up upgrading gst-base-plugins to version release 1.22.10.

This was never an issue with QGroundcontrol, but instead with ```glcolorconverter``` plugin.  Can easily be resolved by taking the following steps:

```
sudo add-apt-repository ppa:savoury1/multimedia
sudo apt-get update
sudo apt-get dist-upgrade
```

booo, 2024-04-03:
Thanks for the feedback!

closed 2024-04-03 by booo