mavlink/qgroundcontrol #10564: Support viewing `named_float` in instrument panel.

eduardkieser, 2023-02-08:
I have implemented a sensor driver in lua on ardupilot for an unsupported temperature sensor. I would like to view the values of that sensor in the "Instrument Panel". The message is being sent as a `named_float` from the scripting environment. On mission planner I can configure the quick view to show the values, but I would prefer to use it in QGC so that I can read the values on the herelink in stead of connecting an additional computer.

On MP's quick view the name of the `named_float` is prepended with `MAV_`, not sure if that helps.

csik, 2023-05-19:
Also interested in this! Great for debugging and breaking in new features/devices!

paularthur, 2024-05-24:
Hi Team - Im liking QGC but lack of support for output from LUA scripts from our ArduRover is pushing us back to Mission Planner
The named_float mechanism on Mission Planners Quickview tab displaying MAV_xxx values is a showstopper for us :( and probably for any other project requiring a small amount of LUA functionality. Are there any workarounds available ?? Thanks in advance.

shift-tyson, 2024-06-06:
QGC does not seem to handle NAMED_VALUE_FLOAT messages, which is kind of surprising.

I'm using a Lua script to send a NAMED_VALUE_FLOAT message to QGC.

I ended up having to add NAMED_VALUE_FLOAT message handling functionality to the Vehicle class.

A good starting point is the switch statement in Vehicle::_mavlinkMessageReceived

HTRamsey, 2024-06-06:
I'll probably hit DEBUG, DEBUG_VECT, NAMED_VALUE_FLOAT, and NAMED_VALUE_INT all at the same time. Any others?

shift-tyson, 2024-06-06:
The Lua script switches vehicle control between two controllers and sends the controller ID (1 Field Control Station or 2 Handheld) via MAVLink NAMED_VALUE_FLOAT message.

The active controller ID is stored in a UI element which displays a gamepad icon or laptop icon in the top right corner of the toolbar. The UI element hooks into the active vehicle's NAMED_VALUE_FLOAT messages via a custom function in the Vehicle class.

h2jaafar, 2025-01-20:
Was this feature ever completed?

HTRamsey, 2025-01-20:
Not yet, but still planning on it

skemp117, 2025-01-21:
I am also looking for something like this> named_value_float is a very easy way to debug flight code. I've been using mavproxy to send named_value_floats, but it will not display them

HTRamsey, 2025-01-21:
Do you guys have an idea of how you would like them displayed? Right now you can at least see it in mavlink inspector

skemp117, 2025-01-21:
For me, it has been a useful tool for working with a companion MCU. I have a Teensy using MAVLink to send and recieve named_value_floats (chosen because of MAVProxy's support for sending namedvaluefloats). It would be really nice to setup custom buttons where you could predefine a message to send and then it is sent when the button is pressed. Same for the receiving. If there was a console or something similar to MAVProxy along the bottom of the screen that you could configure in a similar manner to MAVProxy's "watch" command, that would be awesome! Honestly, even just a list somewhere would be nice. Configurable such that you can have:
name1: 32.1
name2: 54.65

etc...

h2jaafar, 2025-01-21:
Currently for our use case, we are using the following in Mission Planner to visualize custom sensor integration and estimation techniques, where we send named floats over mavlink from our processing algorithm for pilots to view in the field

[image]

skemp117, 2025-01-21:
Yeah, I've also done that. It is just frustrating that it needs to wait for a message before it is able to display it. I'd like to be able to upload a list of names. I've also had problems when I reload mission planner, they will appear as "CustomField1" and I need to rebind them. I also wish you could SEND mavlink messages in MP or QGC like you can in MAVProxy.

paularthur, 2025-01-21:
Thanks for progressing this, as mentioned above for us, this is a showstopper for using QGC and has been holding us back.

Mission Planner allows displaying MAV_xxx in the same way as any other variable, ideally QGC would allow chosing MAV_xxx in the same way.

Key driver is  LUA script output and feedback. Any mechanism to make this easy with QGC is a win !

HTRamsey, 2025-01-21:
Alright sounds good I'll try and get at least something minimal in there in the next couple of weeks

h2jaafar, 2025-01-22:
Yes Lua feedback is the main thing this would be very helpful with

Tensho, 2025-01-23:
A similar `MAV_` name prefix for the named float value message would be great to have in QGC 👍

Tensho, 2025-01-23:
BTW, does the latest ArduPlane SITL (4.5.5) forward messages to Mission Planner automatically now?  [This PR](https://github.com/ArduPilot/MissionPlanner/pull/3226) should make this work, but I don't see the value changes in a "Quick" tab.

Larry84, 2025-01-30:
I would also like to have this feature, as in QGC there is currently no way to display data from multiple custom sensors without complex programming or workarounds. @HTRamsey thank you for taking the time to add this.

skemp117, 2025-02-11:
https://github.com/Persimia/NamedValueFloatSupport.git

Here is a solution I made for Mission Planner. It works well enough for me, but I'd be happy if someone who knows more about Windows Form layouts made it dynamically scale a little better. I was getting too frustrated with it lol.