mavlink/qgroundcontrol #10568: QGroundCOntrol build Failed

SebastianViollaz, 2023-02-10:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Succesfull build in QT
## Current Behavior
/qgroundcontrol/libs/libevents/libevents/libs/cpp/parse/parser.h:13: error: nlohmann_json/include/nlohmann/json_fwd.hpp: No such file or directory
In file included from /qgroundcontrol/libs/libevents/libevents/libs/cpp/parse/parser.cpp:1:
/qgroundcontrol/libs/libevents/libevents/libs/cpp/parse/parser.h:13:10: fatal error: nlohmann_json/include/nlohmann/json_fwd.hpp: No such file or directory
   13 | #include "nlohmann_json/include/nlohmann/json_fwd.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.04.1
- QGC Version: Last in GitHub
- QGC build: FronSource

## Detailed Description
I try to open de .pro file in QT Creator, when i try to build this error show up. The path is not an existing one

## Log Files and Screenshots
[image]

TheLastImperial, 2023-02-12:
In my case the `nlohmann_json` of `libevents` was not there. I just go to `libs/libevents/libevents` and run:

```bash
git submodule init
git submodule update
```

urmyfaith, 2023-02-12:
> In my case the submodule `nlohmann_json` of `libevents` was not there. I just go to `libs/libevents/libevents` and run:
> 
> ```shell
> git submodule init
> git submodule update
> ```

nice

booo, 2023-02-14:
Please provide us with the exact steps to reproduce the problem.

Please check the documentation for a howto.

SebastianViollaz, 2023-02-14:
> In my case the submodule `nlohmann_json` of `libevents` was not there. I just go to `libs/libevents/libevents` and run:
> 
> ```shell
> git submodule init
> git submodule update
> ```

this solve the problem for me

booo, 2023-02-14:
Ok. I close this. Seems to be human error ;)

closed 2023-02-14 by booo