mavlink/qgroundcontrol #10570: Does QGC support other special referance systems for maps?

AndKe, 2023-02-13:
If a WMTS server delivers ESPG:25833 tiles  - should QGC be able to use those, or is EPSG:4326 mandatory?
- I see no place to specify this.

booo, 2023-02-14:
I think you have to provide QGC with the right tile format.

AndKe, 2023-02-14:
can it be that if those two functions did calculate for ESPG:25833  - it would work?   (adding an alternate calculation for given map)
https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/MapProvider.cpp#L80  https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/MapProvider.cpp#L85

booo, 2023-02-14:
Sorry, I'm not familiar with this part of the codebase. Give it a try and let us know.

I'm not sure how the conversion from screen pixels to coordinates works and maybe you need to do some transformation there too?

AndKe, 2023-02-14:
@booo I think that's too deep water for me, as you say - the displaying-part may get wonky too.

closed 2023-02-14 by AndKe