mavlink/qgroundcontrol #10571: Opening QGroundControl in a Docker Container

arongu321, 2023-02-14:
Is there anyway I can run QGroundControl.AppImage inside a docker container as I'm not allowed to run it because I'm a root user in the docker container?

closed 2023-02-14 by booo

arongu321, 2023-02-14:
I fixed it now thanks.

booo, 2023-02-14:
How?

arongu321, 2023-02-14:
So what I had to do was once I installed the modules and packages in the Dockerfile I added the user using
```
RUN useradd -ms /bin/bash aron
USER aron
```
Once I did that I had to install python module PySide2 using
```
RUN pip3 install --user PySide2
```
Here what I did was first open the docker container as the current user aron using:
```
sudo docker run -it --env="DISPLAY" -v /tmp/.X11-unix:/tmp/.X11-unix --privileged --name vicki_container vicki_ros bash
```
Then I was able to navigate to where QGroundControl.AppImage was and managed to successfully open up QGroundControl.AppImage that way.

booo, 2023-02-15:
Cool. Thanks for documenting here.

bigbellmercy, 2024-03-08:
Hi, in my case, following error happens:
`fuse:` failed to exec fusermount: No such file or directory
or
`fuse:` failed to exec fusermount: No such file or directory
Is there some solution?

celikemir, 2024-06-21:
> Hi, in my case, when I run QGroundControl, following error happens: `fuse:` failed to exec fusermount: No such file or directory or `fuse:` failed to exec fusermount: No such file or directory Is there some solution?

run container with --privileged