mavlink/qgroundcontrol #10573: UDP/TCP Conection issue.

SebastianViollaz, 2023-02-14:
----
I use:

- Ubuntu 22.04
- The last one on QGround Repository (4.2.0)

Issue:

- I want to make an API that get data from QGC , battery status, gps, etc. I what QGC to send data via port to that API.
- I made a simple code in c++ that listen a port  and print in console the info that is in the buffer. Let say that this program is mi server and QGC mY client.
-In QGC i made this connection:
[image]
When i start the server it conect to the QGC port  that is in the picture. But in the server side the data is not showing.
-The server work properly ( i tried "telnet 127.0.0.1 port" and it works). 

 I read something about MAVSDK and MavLink Router, bu not sure if that whill help. I understand that this two software are from connecting Ground Stations to an UAV. I made my server without QT Frame work. Is that a problem? 

Server Code: 

I am missing something? How Could i fix this problem?

booo, 2023-02-15:
Please use tcpdump or wireshark to check if your configuration actually tries to establish a connection to the server. Maybe try manual connection first.

SebastianViollaz, 2023-02-15:
I tried manually connect to the server whit the telnet line command in ubuntu. 
Like this: 
[image]
I send data from the terminal, and the server prints it, when i connect to QGC the connection is recived. (I know because a message shows up when the connection is recived), but the data that QGC send doesn't show. 

If i establish the connection for like 4-5 minutes some data shows, then stops printing.:
[image]
this is the showed data in the VSC terminal.

This isnt't the way that supposed to work, it sholud be printing data constantly.

NOTE: The data showed prints like unknown charaters, i read the source code and i thinks that is because it send data of type ,"QByteArray". This is the source code:

[image]

booo, 2023-02-15:
Please follow my suggestion. Wireshark has even a mavlink parser to check the packets.

I don't think this is a problem related to QGC. It's more a problem of your server.

You can use netcat to create a server too. Maybe you should try that as well.

DonLakeFlyer, 2023-02-18:
You are doing the wrong thing. A Comm Link is for communication to/from a vehicle. It is not to forward mavlink. To forward mavlink go to MAVLink page and turn on Enabled MAVLink forwarding and set your host.

closed 2023-02-18 by DonLakeFlyer