mavlink/qgroundcontrol #10574: Hypersensitive scrolling in "Application settings - > General"

AndKe, 2023-02-14:
## Expected Behavior
Expecting to be able to scroll thru "general" settings in a reasonable way (people are usually looking for something here, no need to speed up)

## Current Behavior
Unless scrolling very carefully, it enters "warp speed"  :) and ends instantly at the top/end of the page.

## Steps to Reproduce:
scroll on said page with a mouse, or even worse: the side of a touchpad.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 22.10
- QGC Version: master
- QGC build: n/a

## Detailed Description
this is clearly something with an acceleration setting for scrolling, no other application reacts like this.
- scrolling to zoom in the map is done at a perfectly fine rate.

## Log Files and Screenshots

DonLakeFlyer, 2023-02-18:
Wow, that is pretty crazy!

DonLakeFlyer, 2023-02-18:
@AndKe Can you try this out when it goes through. I set it to 2 inches per sec. It's fairly slow but I think it works.

AndKe, 2023-02-19:
@DonLakeFlyer I tried it, it proved a bit slow, increased it to 8 (inch/s) which still prevents the crazy high accelerations using touchpads and MX mouses, and still allows a decent scrolling speed at which user can easily spot the setting he is looking for.

Tested on various screen resolutions and with following HID devices: (MX mouse, touchpads HP/Dellwith two-finger scrolling and side-scrolling)

Please increase to 8 ```  maximumFlickVelocity:   (ScreenTools.realPixelDensity * 25.4) * 8   ```
Then the commit solves the issue perfectly.
Thank you.

AndKe, 2023-02-19:
also: @DonLakeFlyer  - what is your nickname on Discord? - I'd like to ask you something on ArduPilot/qgroundcontrol ..

DonLakeFlyer, 2023-02-19:
> nickname on Discord?

Don Gagne#4187

DonLakeFlyer, 2023-02-19:
> Please increase to 8

Done

AndKe, 2023-02-19:
@DonLakeFlyer Thank you, well done, now we just wait for the overlords of pulling to pull our PR's - maybe a sacrifice can speed things up :)

closed 2024-05-31 by HTRamsey