mavlink/qgroundcontrol #10577: Access for maintenance

booo, 2023-02-15:
Hi,

I would like to request access to the repository to help maintain the project. For some time now I try to organize the tickets and pull requests and answer questions that pop up. I still don't understand the complete project and all it's processes but I'm interested to learn more and help out. I have a strong computer science background and worked with open source projects in the past.

I work for an association called [SearchWing e.V.](https://searchwing.org) as UAV developer and we make heavy use of the open source/free software ecosystem QGC is part of and we would like to contribute as much as we can to the ecosystem.

I would like to get some feedback from the "old" devs (@patrickelectric, @mrpollo, @DonLakeFlyer and others) on the topic and in the best case permission to maintain the project even more than I do already.

If you have any questions about me or our project please ask!

Best Regards
Philipp

julled, 2023-02-15:
I am also part of the SearchWing team and can support on maintaining QGC.

mrpollo, 2023-02-15:
Great to read, why don you join the Bi-weekly call and we can get organized?

https://dronecode.org/calendar

booo, 2023-02-15:
> Great to read, why don you join the Bi-weekly call and we can get organized?
> 
> https://dronecode.org/calendar

I will try to join the next call. I joined the discord server too. Feel free to ping me if you have any questions.

Davidsastresas, 2023-02-19:
I would love @booo to have access, he has been helping a lot on the recent past. Thank you!

DonLakeFlyer, 2023-02-19:
In order to do releases @booo will need access to the PX4 S3 as well.

mrpollo, 2023-02-19:
When we have our call we can discuss access, happy to help

booo, 2023-02-25:
Access to the repository was granted. We will figure the S3 stuff out while coordinating a release.

Thanks for the trust. I hope I can make a small contribution over time.

closed 2023-02-25 by booo